dotnet/iot #324: Native shim and EINTR handling

tmds, 2019-03-11:
The code relies on specific structs and constants. For example:
```
internal struct epoll_event
{
    public PollEvents events;
    public epoll_data data;
}
```
This works on arm64 and arm32, but this is not ok for x64. On x64, epoll_event is packed so its size is 12 bytes instead of 16.

None of the system calls are handing EINTR. EINTR happens when a system call is interrupted for handing a signal. For example, `epoll_wait` should check for this and retry on EINTR.

corefx handles these thing in a native shim library.

joshfree, 2019-03-11:
@buyaa-n PTAL

joperezr, 2019-04-17:
Our initial supported runtimes are arm32, arm64 and musl arm64, but we will eventually need support in x86 and x64 as well.

tmds, 2019-04-18:
The epoll_event is only an issue for x64.

The EINTR handling applies to all architectures.

joperezr, 2019-07-09:
This isn't required for 3.0 so I'm moving it to future for now. That said, it is definitely a nice to have, specifically the EINTR part.

Tragetaschen, 2019-12-09:
I would argue that the EINTR part is not nice to have, but quite required:
```
System.AggregateException: One or more errors occurred. (Error while waiting for event, error code 4, pin: 8)
 ---> System.IO.IOException: Error while waiting for event, error code 4, pin: 8
   at System.Device.Gpio.Drivers.LibGpiodDriverEventHandler.<>c__DisplayClass12_0.<InitializeEventDetectionTask>b__0()
   at System.Threading.Tasks.Task.InnerInvoke()
   at System.Threading.Tasks.Task.<>c.<.cctor>b__274_0(Object obj)
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location where exception was thrown ---
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot, Thread threadPoolThread)
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
   at System.Threading.Tasks.Task.Wait()
   at System.Device.Gpio.Drivers.LibGpiodDriverEventHandler.Dispose()
   at System.Device.Gpio.Drivers.LibGpiodDriver.Dispose(Boolean disposing)
   at System.Device.Gpio.GpioController.Dispose(Boolean disposing)
   at System.Device.Gpio.GpioController.Dispose()
```

I sometimes see this when ending (CTRL+C) my application, which faithfully calls dispose on everything. The SIGINT results in the system call returning EINTR.

https://github.com/dotnet/iot/blob/1.0/src/System.Device.Gpio/System/Device/Gpio/LibgpiodDriverEventHandler.Linux.cs#L59

Ellerbach, 2021-07-02:
Is this issue still valid?

tmds, 2021-08-17:
Yes.

krwq, 2021-08-17:
can you provide some repro where we could test this? (ok if it repros only from time to time)

tmds, 2022-11-21:
Fixed by https://github.com/dotnet/iot/pull/1499.

closed 2022-11-21 by tmds