dotnet/iot #330: GPIO pin getting/setting performance chokepoints

JeremyKuhne, 2019-03-13:
Creating a summary issue to hold the data I turn up as I investigate the places where performance can be improved when setting & getting GPIO pins.

There is ample opportunity to improve, partially through optimizations of existing code and partially by optimizing special paths (such as getting/setting multiple values at once).

I'm fixing most of these as I investigate, will break out issues that I don't tackle.

JeremyKuhne, 2019-03-13:
One relatively easy place to improve is with pin caching in suboptimal collections. `GpioController`, for example, uses `HashSet<int>` to track open pins. This would be much better served by using `BitArray` (which is able to handle an arbitrary number of bits). Here is a test exploring the performance difference:

``` C#
    public class BitArrayTest
    {
        private BitArray _bitArray;
        private HashSet<int> _hashSet;

        [Params(0, 1, 8, 16, 32)]
        public int StoredValues;

        [GlobalSetup]
        public void Setup()
        {
            _bitArray = new BitArray(32);
            _hashSet = new HashSet<int>();
            for (int i = 0; i < StoredValues; i++)
            {
                _bitArray[i] = true;
                _hashSet.Add(i);
            }
        }

        [Benchmark(Baseline = true)]
        public bool HashSet() => _hashSet.Contains(0) || _hashSet.Contains(15) || _hashSet.Contains(31);

        [Benchmark]
        public bool BitArray() => _bitArray[0] || _bitArray[15] || _bitArray[31];
    }
```

And the results:

|   Method | StoredValues |     Mean |     Error |    StdDev |   Median | Ratio | RatioSD |
|--------- |------------- |---------:|----------:|----------:|---------:|------:|--------:|
|  **HashSet** |            **0** | **4.414 ns** | **0.3457 ns** | **0.8349 ns** | **3.980 ns** |  **1.00** |    **0.00** |
| BitArray |            0 | 3.875 ns | 0.4407 ns | 0.8277 ns | 3.476 ns |  0.83 |    0.11 |
|          |              |          |           |           |          |       |         |
|  **HashSet** |            **1** | **9.102 ns** | **0.1430 ns** | **0.1338 ns** | **9.116 ns** |  **1.00** |    **0.00** |
| BitArray |            1 | 1.783 ns | 0.0893 ns | 0.0836 ns | 1.773 ns |  0.20 |    0.01 |
|          |              |          |           |           |          |       |         |
|  **HashSet** |            **8** | **9.082 ns** | **0.1980 ns** | **0.1654 ns** | **9.074 ns** |  **1.00** |    **0.00** |
| BitArray |            8 | 1.532 ns | 0.0758 ns | 0.0709 ns | 1.539 ns |  0.17 |    0.01 |
|          |              |          |           |           |          |       |         |
|  **HashSet** |           **16** | **9.035 ns** | **0.2061 ns** | **0.1928 ns** | **9.159 ns** |  **1.00** |    **0.00** |
| BitArray |           16 | 1.760 ns | 0.0645 ns | 0.0571 ns | 1.764 ns |  0.20 |    0.01 |
|          |              |          |           |           |          |       |         |
|  **HashSet** |           **32** | **8.997 ns** | **0.1630 ns** | **0.1525 ns** | **8.990 ns** |  **1.00** |    **0.00** |
| BitArray |           32 | 1.634 ns | 0.0803 ns | 0.0788 ns | 1.643 ns |  0.18 |    0.01 |

Checking three values in an "empty" collection is slightly faster with `BitArray`. Checking three values when any number of values are stored is **5x** faster (which should be the typical case, it will be unlikely that there are no open pins).

joperezr, 2019-04-16:
@JeremyKuhne is this still something you have plans on doing? If not let me know so that we can flag this as up-for-grabs instead.

JeremyKuhne, 2019-04-16:
> is this still something you have plans on doing

I want to, but anyone is welcome to take this on if they can allocate time before I can.

joperezr, 2019-05-14:
Also, it would be great if we can at least classify which classes to focus on first. That way we may be able to split this issue into smaller ones that can be planned in or out of 3.0 release.

joshfree, 2019-06-11:
Moving to Future milestone and marking as up-for-grabs

Ellerbach, 2021-07-08:
[Triage] related to #1490

krwq, 2023-02-09:
[Triage] Closing as duplicate of of https://github.com/dotnet/iot/issues/1490

closed 2023-02-09 by krwq