dotnet/iot #336: NuGet missing package Error updating beyond 191156.1

PRIMETSS, 2019-03-17:
Trying to upgrade Iot.DeviceBindings to anything version above 0.1.0-prerelease.19156.1 get this error in NuGet Manager

Installing System.Device.Gpio 0.1.0-prerelease.19158.2.
Installing Iot.Device.Bindings 0.1.0-prerelease.19158.2.
NU1101: Unable to find package Microsoft.Net.Compilers.Toolset. No packages exist with this id in source(s): aspnetcore-release, CliFallbackFolder, DotNetCore ASPNET Dev Builds, DotnetCore Dev Builds, Microsoft Visual Studio Offline Packages, NuGet, NuGet v2
Package restore failed. Rolling back package changes for 'DT80Tester'.
Time Elapsed: 00:00:03.1361678

Using standar NuGet repos plus https://dotnet.myget.org/F/dotnet-core/api/v3/index.json

alexdresko, 2019-03-20:
Just wanted to confirm that I hit the same problem.

krwq, 2019-03-21:
@safern, is this perhaps related to the recent compiler update? seems like this package should be build only dependency

safern, 2019-03-21:
Yep. I’ll fix it tomorrow.

closed 2019-03-21 by safern