dotnet/iot #340: Use System.Device.Gpio in Powershell Core 6

Kraftmuschel, 2019-03-20:
Hi,

would like to use System.Device.Gpio in Powershell Core but it's not working. 

I installed latest Windows IoT on a Raspberry Pi 3b, the DotNet prerelease and Powershell Core 6. To be able to install the System.Device.Gpio assembly I created a new project in DotNet. Then I used the documented command to install the DLL finally. In C# the commands are available. 

In Powershell I used 

PS C:\> Add-Type -AssemblyName System.Device.Gpio

to load the System.Device.Gpio assembly successfull - at least Powershell does not return an error message. However the command

PS C:\> [System.Device.Gpio] | Get-Member 

does not work.

This might be a stupid question, but is the assembly expected to work in Powershell?

Kind regards

krwq, 2019-03-22:
Windows IoT has not been thoroughly tested yet (Raspbian has been the mostly focues) - I'd expect assembly loading to work regardless of the current state of this repo.

Currently Powershell is marked as experimental on both Windows and Raspbian (see README page):
https://github.com/powershell/powershell

I believe that would be the best place to ask the question. I'd recommend to open issue over there and write specific commands you are using (if they use external files please provide where you got that from) and write down the output powershell produces.

If they confirm this is a powershell issue then we can close this issue if there is some issue in here we would need some more details.

Also you might want to try with the new package we have produced today:
https://www.nuget.org/packages/System.Device.Gpio/0.1.0-prerelease.19171.3

(assuming you were using myget before you might have hit this issue: https://github.com/dotnet/iot/issues/336 but it is really hard to help without any error message you are getting)

joshfree, 2019-04-02:
@joperezr PTAL

Kraftmuschel, 2019-04-03:
Hello,

Thank you for your help. Today I tried again with the latest version. I used the command

dotnet add package System.Device.Gpio -Version 0.1.0-prerelease-19171.3 

to install the package.

I used 

find . -Name System.Device.Gpio.dll 

to locate the DLL on my device. Surprisingly, I found the file in a few directories, for example:

./.nuget/packages/system.device.gpio/0.1.0-prerelease.19171.3/runtimes/linux/lib/netstandard2.0/System.Device.Gpio.dll
./.nuget/packages/system.device.gpio/0.1.0-prerelease.19171.3/runtimes/win/lib/netstandard2.0/System.Device.Gpio.dll

but the DLL is not available in the Directory …/linux-arm/

Do I need a special version for the Raspberry Pi?

Kind regards

krwq, 2019-04-03:
@Kraftmuschel `linux` is compatible with `linux-arm`. `linux-arm` specific binaries should be visible when you publish the runtime assets (i.e. `dotnet publish -r linux-arm`) - that should be at minimum `.so` file

Kraftmuschel, 2019-04-04:
Hi, 

I found the file System.Device.Gpio.Native.So

/home/pi/.nuget/packages/system.device.gpio/0.1.0-prerelease.19171.3/runtimes/linux-arm/native
System.Device.Gpio.Native.so

but no DLL in this tree. So I loaded the DLL from the linux subfolder

PS /home/pi/.nuget/packages/system.device.gpio/0.1.0-prerelease.19171.3/runtimes/linux/lib/netstandard2.0> Add-Type -AssemblyName ./System.Device.Gpio.dll   

This command does not return any error message. However, when I try to get the members, it fails:

PS /home/pi/.nuget/packages/system.device.gpio/0.1.0-prerelease.19171.3/runtimes/linux/lib/netstandard2.0> [System.Device.Gpio] | Get-Member                    Unable to find type [System.Device.Gpio].                                       At line:1 char:1
+ [System.Device.Gpio] | Get-Member
+ ~~~~~~~~~~~~~~~~~~~~
+ CategoryInfo          : InvalidOperation: (System.Device.Gpio:TypeName) [], RuntimeException
+ FullyQualifiedErrorId : TypeNotFound

Kind regards

krwq, 2019-04-04:
note that System.Device.Gpio is a namespace not a type (that's what the error is about). Try i.e. `System.Device.Gpio.GpioController` instead

Kraftmuschel, 2019-04-05:
Thank you for getting this straigth. 

[System.Device.Gpio.GpioController] | Get-Member

Works instantly. I will continue my attempts to make use of it in Powershell this weekend.

joperezr, 2019-04-15:
Closing as it seems like the issue here is resolved. @Kraftmuschel please reopen if that is not the case.

closed 2019-04-15 by joperezr