dotnet/iot #345: .NET Core 3.0 Raspberry Pi 3 Install - FileNotFoundException

IVData, 2019-03-24:
Apologies if this is not the right place but I feel that I'm missing something simple.

What I'm trying to do is test out razor components and .NET core 3.0 with System.Device.Gpio, developing on Windows in visual studio and running the app on a raspberry pi. I have been following [this guide](https://docs.microsoft.com/en-us/aspnet/core/razor-components/get-started?view=aspnetcore-3.0&tabs=visual-studio) to get started.

I have installed the .NET core 3.0 SDK on the raspberry pi, mounted the solution folder and run `dotnet restore` and `dotnet run` - the default project works great.

However when I add System.Device.Gpio library from here, the runtime on the pi is unable to find the library (I ran `dotnet restore` again) and throws an exception as soon as I try to load a page.
```
System.IO.FileNotFoundException: Could not load file or assembly 'System.Device.Gpio, Version=0.1.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35'. The system cannot find the file specified.

File name: 'System.Device.Gpio, Version=0.1.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35'
   at System.Reflection.RuntimeAssembly.nLoad(AssemblyName fileName, String codeBase, RuntimeAssembly locationHint, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, IntPtr ptrLoadContextBinder)
   at System.Reflection.RuntimeAssembly.InternalLoadAssemblyName(AssemblyName assemblyRef, RuntimeAssembly reqAssembly, StackCrawlMark& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, IntPtr ptrLoadContextBinder)
   at System.Reflection.Assembly.Load(AssemblyName assemblyRef)
   at Microsoft.AspNetCore.Components.ComponentResolver.EnumerateAssemblies(AssemblyName assemblyName, Assembly blazorAssembly, HashSet`1 visited)+MoveNext()
   at System.Linq.Enumerable.SelectManySingleSelectorIterator`2.MoveNext()
   at Microsoft.AspNetCore.Components.ComponentResolver.EnumerateAssemblies(AssemblyName assemblyName, Assembly blazorAssembly, HashSet`1 visited)+MoveNext()
   at System.Linq.Enumerable.SelectManySingleSelectorIterator`2.MoveNext()
   at System.Linq.Enumerable.WhereEnumerableIterator`1.MoveNext()
   at Microsoft.AspNetCore.Components.Routing.RouteTable.Create(IEnumerable`1 types)
   at Microsoft.AspNetCore.Components.Routing.Router.SetParametersAsync(ParameterCollection parameters)
   at Microsoft.AspNetCore.Components.Rendering.ComponentState.SetDirectParameters(ParameterCollection parameters)
   at Microsoft.AspNetCore.Components.RenderTree.RenderTreeDiffBuilder.InitializeNewComponentFrame(DiffContext& diffContext, Int32 frameIndex)
   at Microsoft.AspNetCore.Components.RenderTree.RenderTreeDiffBuilder.InitializeNewSubtree(DiffContext& diffContext, Int32 frameIndex)
   at Microsoft.AspNetCore.Components.RenderTree.RenderTreeDiffBuilder.InsertNewFrame(DiffContext& diffContext, Int32 newFrameIndex)
   at Microsoft.AspNetCore.Components.RenderTree.RenderTreeDiffBuilder.AppendDiffEntriesForRange(DiffContext& diffContext, Int32 oldStartIndex, Int32 oldEndIndexExcl, Int32 newStartIndex, Int32 newEndIndexExcl)
   at Microsoft.AspNetCore.Components.RenderTree.RenderTreeDiffBuilder.ComputeDiff(Renderer renderer, RenderBatchBuilder batchBuilder, Int32 componentId, ArrayRange`1 oldTree, ArrayRange`1 newTree)
   at Microsoft.AspNetCore.Components.Rendering.ComponentState.RenderIntoBatch(RenderBatchBuilder batchBuilder, RenderFragment renderFragment)
   at Microsoft.AspNetCore.Components.Rendering.Renderer.RenderInExistingBatch(RenderQueueEntry renderQueueEntry)
   at Microsoft.AspNetCore.Components.Rendering.Renderer.ProcessRenderQueue()
   at Microsoft.AspNetCore.Components.Rendering.Renderer.AddToRenderQueue(Int32 componentId, RenderFragment renderFragment)
   at Microsoft.AspNetCore.Components.RenderHandle.Render(RenderFragment renderFragment)
   at Microsoft.AspNetCore.Components.ComponentBase.StateHasChanged()
   at Microsoft.AspNetCore.Components.ComponentBase.CallOnParametersSetAsync()
   at Microsoft.AspNetCore.Components.ComponentBase.RunInitAndSetParametersAsync()
```

Is there something I am missing here? I have tried re-installing the library on the pi, and it still throws this error. I feel like this is something simple =\

Thanks

krwq, 2019-03-25:
Hey @IVData,

I do not have any expertise on the Razor components but perhaps I will be able to help or redirect to the correct place.

Could you share you csproj (or at least all PackageReference and possible conditional related to them)?

Also please let us know how do you restore and build.

Also please share `dotnet --info` output.

Here is what you can also try (on your Windows box):
`dotnet publish -r linux-arm`

make sure to run that on your exe project (directory which contains exe's csproj)
and then copy whole `./bin/Debug/<framework>/linux-arm/publish/` (`<framework>` - i.e. netcoreapp2.1) to your Raspberry Pi.

On your Raspberry Pi `chmod +x` the executable (it usually has the same name as the folder or csproj file) and try to run it. (`+x` is needed because Windows does not have the notion of such permissions so it will be lost when copying)

This should not require anything installed on Raspberry Pi (except perhaps some external dependencies) and should just work.

IVData, 2019-03-25:
Hi,

I'll test the publishing when I get home but in the mean time I have the csproj and ran `dotnet --info`.

`dotnet --info`:
```
.NET Core SDK (reflecting any global.json):
 Version:   3.0.100-preview3-010431
 Commit:    d72abce213

Runtime Environment:
 OS Name:     raspbian
 OS Version:  9
 OS Platform: Linux
 RID:         linux-x64
 Base Path:   /home/pi/dotnet/sdk/3.0.100-preview3-010431/

Host (useful for support):
  Version: 3.0.0-preview3-27503-5
  Commit:  3844df9537

.NET Core SDKs installed:
  3.0.100-preview3-010431 [/home/pi/dotnet/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.App 3.0.0-preview3-19153-02 [/home/pi/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 3.0.0-preview3-27503-5 [/home/pi/dotnet/shared/Microsoft.NETCore.App]
```

csproj:
```
<Project Sdk="Microsoft.NET.Sdk.Web">

  <PropertyGroup>
    <TargetFramework>netcoreapp3.0</TargetFramework>
    <LangVersion>7.3</LangVersion>
    <_RazorComponentInclude>Components\**\*.cshtml</_RazorComponentInclude>
  </PropertyGroup>

  <ItemGroup>
    <PackageReference Include="IoT.Device.Bindings" Version="0.1.0-prerelease.19172.1" />
    <PackageReference Include="Microsoft.AspNetCore.Components.Server" Version="3.0.0-preview3-19153-02" />
    <PackageReference Include="Microsoft.AspNetCore.Mvc.NewtonsoftJson" Version="3.0.0-preview3-19153-02" />
    <PackageReference Include="System.Device.Gpio" Version="0.1.0-prerelease.19172.1" />
  </ItemGroup>

</Project>
```

I was trying to avoid publishing if I can, since I'm developing on another machine I'd like to run `dotnet watch run` on the raspberry pi while I dev to speed up the process of trying things. But I'll give it a go as it may lead us somewhere.

> or when System.Device.Gpio runtime assembly did not get copied to the PI

I would have thought since my setup has the whole solution folder on the Pi, that removing and re-installing the packages from the Pi would have made the correct references. But maybe not.

IVData, 2019-03-28:
Finally had time to try publishing - and it works as expected.
Some linux-arm versions of `System.Device.Gpio` are definitely present in the project dir:

```
./bin/Debug/netcoreapp3.0/linux-arm/System.Device.Gpio.Native.so
./bin/Debug/netcoreapp3.0/linux-arm/publish/System.Device.Gpio.Native.so
./bin/Debug/netcoreapp3.0/publish/System.Device.Gpio.Native.so
./bin/Debug/netcoreapp3.0/runtimes/linux-arm/native/System.Device.Gpio.Native.so
./obj/Debug/netcoreapp3.0/linux-arm/PubTmp/Out/System.Device.Gpio.Native.so
```

So I'm still not sure why `dotnet run` is unable to find it.

krwq, 2019-03-28:
@IVData I think I have an idea what the issue might be.

Your `dotnet --info` on RPi shows `RID: linux-x64` - this seems incorrect.
it should have printed `linux-arm` - how did you install .NET Core on RPi?
is this the ARM32 package/zip from: https://dotnet.microsoft.com/download ?

Note: `System.Device.Gpio.Native` is OOB nuget package (OOB meaning that it doesn't ship with .NET Core and is a seperate package) - that means that you won't have native dependencies unless you restore that package for linux-arm or deploy your app with linux-arm specific binaries (i.e. *.so files).

could you try passing correct runtime to dotnet run? (`dotnet run -r linux-arm`)

richlander, 2019-03-29:
I can repro this with the [led-blink sample](https://github.com/dotnet/iot/tree/master/samples/led-blink) by simply updating to the latest version of the package. I repro'd on both .NET Core 2.1 and 3.0.

```console
pi@raspberrypi:~/led-blink $ cat led-blink.csproj
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>netcoreapp3.0</TargetFramework>
    <RootNamespace>led_blink</RootNamespace>
  </PropertyGroup>

  <ItemGroup>
    <PackageReference Include="System.Device.Gpio" Version="0.1.0-prerelease.19078.2" />
  </ItemGroup>

</Project>
pi@raspberrypi:~/led-blink $ dotnet run
Let's blink an LED!
GPIO pin enabled for use: 17
Light for 1000ms
Dim for 200ms
Light for 1000ms
^C
pi@raspberrypi:~/led-blink $ nano led-blink.csproj
pi@raspberrypi:~/led-blink $ cat led-blink.csproj
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>netcoreapp3.0</TargetFramework>
    <RootNamespace>led_blink</RootNamespace>
  </PropertyGroup>

  <ItemGroup>
    <PackageReference Include="System.Device.Gpio" Version="0.1.0-prerelease.19171.3" />
  </ItemGroup>

</Project>
pi@raspberrypi:~/led-blink $ dotnet build
Microsoft (R) Build Engine version 16.1.27-preview+g91af6445a6 for .NET Core
Copyright (C) Microsoft Corporation. All rights reserved.

  Persisting no-op dg to /home/pi/led-blink/obj/led-blink.csproj.nuget.dgspec.json
  Restore completed in 1.07 sec for /home/pi/led-blink/led-blink.csproj.
/usr/share/dotnet/sdk/3.0.100-preview4-010965/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.RuntimeIdentifierInference.targets(151,5): message NETSDK1057: You are using a preview version of .NET Core. See: https://aka.ms/dotnet-core-preview [/home/pi/led-blink/led-blink.csproj]
  led-blink -> /home/pi/led-blink/bin/Debug/netcoreapp3.0/led-blink.dll

Build succeeded.
    0 Warning(s)
    0 Error(s)

Time Elapsed 00:00:45.21
pi@raspberrypi:~/led-blink $ dotnet run

Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'System.Device.Gpio, Version=0.1.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35'. The system cannot find the file specified.

pi@raspberrypi:~/led-blink $ dotnet --info
.NET Core SDK (reflecting any global.json):
 Version:   3.0.100-preview4-010965
 Commit:    91b94bcdef

Runtime Environment:
 OS Name:     raspbian
 OS Version:  9
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /usr/share/dotnet/sdk/3.0.100-preview4-010965/

Host (useful for support):
  Version: 3.0.0-preview4-27525-12
  Commit:  6001915623

.NET Core SDKs installed:
  3.0.100-preview-010190 [/usr/share/dotnet/sdk]
  3.0.100-preview4-010763 [/usr/share/dotnet/sdk]
  3.0.100-preview4-010963 [/usr/share/dotnet/sdk]
  3.0.100-preview4-010965 [/usr/share/dotnet/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.App 3.0.0-preview-19075-0444 [/usr/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.0.0-preview4-19164-03 [/usr/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 3.0.0-preview4-19172-03 [/usr/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 3.0.0-preview-27329-4 [/usr/share/dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.0.0-preview4-27514-07 [/usr/share/dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 3.0.0-preview4-27525-12 [/usr/share/dotnet/shared/Microsoft.NETCore.App]

To install additional .NET Core runtimes or SDKs:
  https://aka.ms/dotnet-download
pi@raspberrypi:~/led-blink $
```

krwq, 2019-03-29:
@richlander does it make a difference if you do `dotnet run -r linux-arm` - it seems to me like dotnet cli issue to be honest (doesn't restore runtime assets)

richlander, 2019-03-29:
Yes.

```console
pi@raspberrypi:~/led-blink $ dotnet run -r linux-arm
Let's blink an LED!
GPIO pin enabled for use: 17
Light for 1000ms
Dim for 200ms
Light for 1000ms
Dim for 200ms
Light for 1000ms
Dim for 200ms
```

/cc @nguerrera -- what should happen here so that `dotnet run` works on its own? This is the package: https://www.nuget.org/packages/System.Device.Gpio/0.1.0-prerelease.19171.3

IVData, 2019-03-29:
> how did you install .NET Core on RPi?

I followed the instructions here since I was looking for the 3.0 preview: https://dotnet.microsoft.com/download/dotnet-core/3.0

I have confirmed now that I downloaded the version listed under preview 3,  listed as "ARM32" and followed their instructions to install it. Very weird that it comes up as linux-x64.
```
mkdir -p $HOME/dotnet && tar zxf dotnet-sdk-3.0.100-preview3-010431-linux-arm.tar.gz -C $HOME/dotnet
export DOTNET_ROOT=$HOME/dotnet
export PATH=$PATH:$HOME/dotnet
```

@richlander how did you go about installing 3.0? I see yours comes up as `linux-arm`, but has a different path and also you've got preview 4 (git version?)

richlander, 2019-03-29:
If you want to be one of the "cool kids" (like me), install the SDK from the first column in the builds table @ https://github.com/dotnet/core-sdk. Stuff may break, but the builds are typically <24 hours old.

Here is the original issue on the linux-x64 RID being incorrectly reported: https://github.com/dotnet/cli/issues/10676

nguerrera, 2019-03-30:
Does this work?

`dotnet publish`
`dotnet bin\Debug\netcoreapp3.0\app.dll`

I suspect it will. Locally, I got all of the runtime targets on publish, but not build. cc @peterhuene 

(This should probably be moved to dotnet/sdk, but I don't have enough access to use Github's transfer and my workaround app is down.)

peterhuene, 2019-03-30:
This should be fixed in nightly. I'm at a happy hour with Microsoft employees in Portland, but I can dig up the issue when I get back.

peterhuene, 2019-03-30:
Oh Rich already linked to it.

IVData, 2019-03-30:
Rich mentioned in his first post that the RID was fixed in nightly, but that `dotnet run` still doesn't work as expected. Unfortunately I can't build my project on linux with the nightly (`IApplicationBuilder.UseRouting()` has changed. But that's an ASP thing).

peterhuene, 2019-03-30:
Sorry, I was viewing and responding to the issue on my phone and didn't fully read the entire thread to see there were multiple issues being discussed.

@nguerrera which runtime targets are missing?

krwq, 2019-03-30:
@peterhuene System.Device.Gpio (OOB package) has some native assets on linux-arm which seem to be missing

peterhuene, 2019-04-01:
I don't have a Pi (or other ARM device) ready to repro on, but here's what I get with a simple console app repro:

```
> mkdir repro && cd repro 
> dotnet new console
> dotnet add package System.Device.Gpio --version 0.1.0-prerelease.19171.3
> dotnet publish
> cd bin\Debug\netcoreapp3.0\publish
> tree /f
Folder PATH listing
Volume serial number is D61A-4405
C:.
│   repro.deps.json
│   repro.dll
│   repro.exe
│   repro.pdb
│   repro.runtimeconfig.json
│   
└───runtimes
    ├───linux
    │   └───lib
    │       └───netstandard2.0
    │               System.Device.Gpio.dll
    │               
    ├───linux-arm
    │   └───native
    │           System.Device.Gpio.Native.so
    │           
    └───win
        └───lib
            └───netstandard2.0
                    System.Device.Gpio.dll                 
```

Given how this package is authored, this appears correct.

@nguerrera do you know if the host respects a TFM of `netstandard2.0` when resolving the `System.Device.Gpio.dll` asset at runtime?

nguerrera, 2019-04-01:
It seems to be working for me too now after updating to latest. I believe I was missing the .so on Friday, but I grabbed a later nightly after that. I don't know what version I had before.

> @nguerrera do you know if the host respects a TFM of netstandard2.0 when resolving the System.Device.Gpio.dll asset at runtime?

I don't think it knows anything about TFM compatibility, but rather picks the best one by RID from the list in .deps.json:

```
 "runtimes/linux/lib/netstandard2.0/System.Device.Gpio.dll": {
            "rid": "linux",
            "assetType": "runtime",
            "assemblyVersion": "0.1.0.0",
            "fileVersion": "0.1.19.17103"
          },
          "runtimes/win/lib/netstandard2.0/System.Device.Gpio.dll": {
            "rid": "win",
            "assetType": "runtime",
            "assemblyVersion": "0.1.0.0",
            "fileVersion": "0.1.19.17103"
          },
```

In other words, the netstandard2.0 choice and path is baked in at build time.

monty124, 2019-04-02:
hi there guys
I'm also having this issue, Rpi3 dotnet core 2.2.103, trying to run the bmp280 sample as a start

any ideas on how i can get this working? development is on a win pc using VS Community 2017

`pi@raspberrypi:~/release $ dotnet --info
.NET Core SDK (reflecting any global.json):
 Version:   2.2.103
 Commit:    8edbc2570a

Runtime Environment:
 OS Name:     raspbian
 OS Version:  9
 OS Platform: Linux
 RID:         linux-arm
 Base Path:   /opt/dotnet/sdk/2.2.103/

Host (useful for support):
  Version: 2.2.1
  Commit:  878dd11e62

.NET Core SDKs installed:
  2.2.103 [/opt/dotnet/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.All 2.2.1 [/opt/dotnet/shared/Microsoft.AspNetCore.All]
  Microsoft.AspNetCore.App 2.2.1 [/opt/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 2.2.1 [/opt/dotnet/shared/Microsoft.NETCore.App]

To install additional .NET Core runtimes or SDKs:
  https://aka.ms/dotnet-download`

`<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>netcoreapp2.2</TargetFramework>
    <LangVersion>7.3</LangVersion>
  </PropertyGroup>

  <ItemGroup>
    <PackageReference Include="Iot.Device.Bindings" Version="0.1.0-prerelease.19171.3" />
    <PackageReference Include="System.Device.Gpio" Version="0.1.0-prerelease.19176.1" />
  </ItemGroup>

</Project>
`
`Unhandled Exception: System.IO.FileNotFoundException: Could not load file or assembly 'System.Device.Gpio, Version=0.1.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35'. The system cannot find the file specified.

   at test.Program.Main(String[] args)
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.Start[TStateMachine](TStateMachine& stateMachine)
   at test.Program.Main(String[] args)
   at test.Program.<Main>(String[] args)
Aborted
`
`pi@raspberrypi:~/release $ ls -R ./runtimes/
./runtimes/:
linux  linux-arm  win  win8

./runtimes/linux:
lib

./runtimes/linux/lib:
netstandard2.0

./runtimes/linux/lib/netstandard2.0:
System.Device.Gpio.dll

./runtimes/linux-arm:
native

./runtimes/linux-arm/native:
System.Device.Gpio.Native.so

./runtimes/win:
lib

./runtimes/win/lib:
netstandard2.0

./runtimes/win/lib/netstandard2.0:
System.Device.Gpio.dll

./runtimes/win8:
lib

./runtimes/win8/lib:
netstandard1.3

./runtimes/win8/lib/netstandard1.3:
System.Runtime.WindowsRuntime.dll`

krwq, 2019-04-02:
@monty124 the easiest way to unblock yourself is to run `dotnet publish -r linux-arm` and copy the publish directory from `bin/` to your raspberry PI (make sure to publish project with executable and if building on Windows after copying you need to add `+x` to your output exe)

peterhuene, 2019-04-02:
@nguerrera were you able to run it though?  Both the build and publish output are correct according to the package layout.  If it can't find System.Device.Gpio.dll, then I don't think this `runtimes` layout is supported by the host.

richlander, 2019-04-02:
Maybe @vitek-karas can offer thoughts.

nguerrera, 2019-04-02:
> @nguerrera were you able to run it though?

nguerrera, 2019-04-02:
> @nguerrera were you able to run it though? 

No, I didn't have access to an arm device. I realize now that what I said about missing runtime targets was a red herring. I had copied '*.dll' into my query, which caused the *.so to be excluded and left me seeing 2 runtime targets instead of 3, but there were probably all 3 there in all cases.

I agree that somebody on the host side should look next.

joshfree, 2019-04-02:
> No, I didn't have access to an arm device

@nguerrera we have RPi 3s you can borrow from our team room.  @krwq can assist.

nguerrera, 2019-04-02:
@joshfree Thanks. At this point, though, we believe the build output is correct and I think somebody on the host / runtime side should look next.

monty124, 2019-04-02:
Hi @krwq 
That did work (for fyi here is the full command to publish to a custom directory)
`dotnet publish -r linux-arm -c Release -o release` from within your project directory (for those that are a little newer to this, like me)
> the easiest way to unblock yourself 
So the question remains, what is the proper way? I was **unable** to generate an exe only a dll, and the targeted rid produced a release package of 187 items and ~65Mb! thats a hell of a lot of supporting libraries for a sample project with 7 references :)
also i was originally targeting "Any" which should include linux-arm from within Visual Studio? https://docs.microsoft.com/en-us/dotnet/core/rid-catalog, not too much of a drama to publish using a command line command but 65Mb of items is a bit much i think!
happy to help where I can.

peterhuene, 2019-04-02:
`-r linux-arm` should work because it won't create a `runtimes` folder on build / publish as we know the specific RID being targeted and can copy the assets into a flat structure.

It shouldn't be required to explicitly target the RID in this case and the host should be able to resolve these assets at runtime.

FYI, when you build with a target RID, it defaults to a self-contained application which includes the entire .NET Core runtime; use `--self-contained false` when publishing to create a framework-dependent deployment (does not include the .NET Core runtime assets).

monty124, 2019-04-02:
@peterhuene 
Thats PERFECT! adding the --self-contained false did the trick, 8 items 381Kb.

`dotnet publish -r linux-arm -c Release -o release --self-contained false`

If i can be of further assistance do let me know.
It seems to be not limited to dotnet core 3+ but

vitek-karas, 2019-04-02:
@monty124 if you can easily repro this, can you please run the app (which would fail) with `COREHOST_TRACE=1` set in the environment. And then capture the `stderr` of the process. It will contain lot of info and will contain local file paths and so on about the machine. If it's OK with you, please share that. If it's not OK, please search in it for the `System.Devide.Gpio`, the interesting bits will be around the occurrences of this string. If you could share some of those... it should show where it tried to search for it (and why) and why it didn't find it. Thanks a lot for your help.

monty124, 2019-04-02:
Hi All

further to this, anyone who may be using visual studio to write code and deploy to Rpi, add this to your csproj file 

`  <PropertyGroup>
..
	<RuntimeIdentifiers>linux-arm</RuntimeIdentifiers>
..
  </PropertyGroup>`

this will allow you to target the Rpi arm RID from within VS and no have to publish using command line, make sure you select the target runtime within your publish profile but!

monty124, 2019-04-02:
hi @vitek-karas 
(for anyone else that may need to do)
`> export COREHOST_TRACE=1` 
on the pi then run the faulting app as such
` >dotnet ./test.dll 2> trace.txt`
trace attached
publish settings are an RID of "Portable" and netcoreapp2.2

[trace.txt](https://github.com/dotnet/iot/files/3036191/trace.txt)

vitek-karas, 2019-04-03:
Thanks a lot @monty124 .

I don't actually know what is the correct behavior, but there's definitely a mismatch between the package/SDK and the host in how they treat the `.deps.json`.

The host does this for every package:
* Reads all the RID specific assets from the `runtimeTargets` section (in the case of System.Device.Gpio there are 3, `linux` assembly, `win` assembly and `linux-arm` so).
* picks the most specific RID which matches the current platform and for which there is at least one asset in the package. (in this case it picks `linux-arm`, since that matches the current platform and there's the so asset for it).
* filters out all assets which are not for the selected RID - so in this case it filters out the `win` and `linux` assets. Losing all managed assemblies in the process.

The end result is that we're left with just the `.so`, but no managed assembly, which is what later on fails at runtime.

This is apparently different behavior than the SDK/nuget expects, as it seems the SDK expects the host to pick the best RID for each asset separately. The problem is that there's no good wat to group the assets,, they're all in one section in the `.deps.json`, and going by file name might be dangerous.

Note: This does not apply to non-RID specific assets, so most assemblies which are RID agnostic will work just fine.

Note: Apps published for a specific RID (so non-portable) don't have this problem since they don't have the `runtimeTargets` section. In their case all the assets are already filtered by nuget/SDK and are only written into the `runtime` section which doesn't have any RIDs. This shows that nuget/SDK can somehow group the assets such that it picks the right set in this case.

@nguerrera Do you know what algorithm is used by SDK/Nuget to filter the assets for a specific RID - as apparently the host is not using the exact same algorithm. Specifically how does it group assets such that it know from which "groups" to pick only one.

nguerrera, 2019-04-03:
Interesting. Thanks, @vitek-karas. Adding @nkolev92, @ericstj who may be able to provide a definitive answer.

The algorithm is in nuget. Everything is picked at restore time. The choices are listed in project.assets.json and the sdk honors them. I'm not sure, but I suspect for runtimetargets case (no rid specified), it is only selecting from the subtrees that have tfm alternatives. And for the case with RID, I think it is selecting the best rid fit for each asset type independently. But I'd have to study the code to be sure.

nguerrera, 2019-04-03:
> The problem is that there's no good wat to group the assets,, they're all in one section in the .deps.json, and going by file name might be dangerous.

If the nuget algorithm is per assetType, I think there is enough information to match that without parsing paths. Each runtimeTarget file in the list has an "assetType".

``` js
"System.Device.Gpio/0.1.0-prerelease.19171.3": {
  "runtimeTargets": {
    "runtimes/linux/lib/netstandard2.0/System.Device.Gpio.dll": {
      "rid": "linux",
      "assetType": "runtime",
      "assemblyVersion": "0.1.0.0",
      "fileVersion": "0.1.19.17103"
    },
    "runtimes/win/lib/netstandard2.0/System.Device.Gpio.dll": {
      "rid": "win",
      "assetType": "runtime",
      "assemblyVersion": "0.1.0.0",
      "fileVersion": "0.1.19.17103"
    },
    "runtimes/linux-arm/native/System.Device.Gpio.Native.so": {
      "rid": "linux-arm",
      "assetType": "native",
      "fileVersion": "0.0.0.0"
    }
  }
},
```

ericstj, 2019-04-03:
Correct: NuGet selects independently by asset type. If the host does not that is a bug.

I'm guessing that folks don't often both cross-compile the assembly by RID (SDK doesn't support it) and provide a native asset: usually if folks invent a PAL they do it in one place.  System.Device.Gpio has two, or rather, it has a managed PAL as well as a native helper.  ML.NET might hit this too: can't recall if it cross-compiles any managed components an includes native in the same package.

vitek-karas, 2019-04-08:
Created dotnet/core-setup#5728 to get this fixed in the host.

maloo, 2019-06-10:
@vitek-karas I tried
`    <RuntimeIdentifier>linux-arm</RuntimeIdentifier>`
`    <SelfContained>false</SelfContained>`
as a temporary workaround in VS. But then I only get the managed dll. Do I need some other property set to get the native one (System.Device.Gpio.Native.so)?
Removing SelfContained gives me _all_ files (about 200). And linux-arm + self contained false gives me all I need except the native so. But I do get the System.IO.Ports.Native.so.

monty124, 2019-06-10:
@maloo 
If you've added this to your csproj file

<PropertyGroup> 
.. 
<RuntimeIdentifiers>linux-arm</RuntimeIdentifiers>
 .. 
</PropertyGroup>

and then selected linux-arm in your publish settings you should get files you need. I just checked my publish settings and output and its working.

maloo, 2019-06-10:
@monty124 
Yes, that's what I've used the last couple of months. But then ssh-deploy have about 200 files to copy over to my Pi at every deploy. So I wanted to limit this (using .Net core installed on the Pi) by adding SelfContained=false. And it worked. I just fixed a bug in the Iot package and temporarily used direct dll references to my local build instead of the nuget. But I missed adding the .so ... so all is fine, my mistake, SelfContained=false works fine until preview7 is released with the real fix. Now I think I prefer this over the portable solution. I just used portable because of the deployment size, RID+SC=false is the best of both worlds.

krwq, 2019-06-11:
another way to limit the transfer is to use `rsync` which copies incrementally

maloo, 2019-06-11:
@krwq
Do you have a PostBuild event example (to paste in csproj)? Is rsync part of Win10? I use ssh-deploy in VS now and it has no incremental support. I really hope VS will add built in ssh deploy and F5 deploy+remote debug by Core 3.0 release ;) . Just like you have for C++.

joperezr, 2019-06-11:
It is worth noting that we will be removing the need of System.Device.Gpio.Native.so from our package, since we will now PInvoke directly into libgpiod.so. When we do that, then this problem will go away.

krwq, 2019-06-11:
@maloo it's installed on linux by default, I think WSL has it, unfortunatelly I only wrote script which builds csproj and then runs rsync:

```
rsync -r ./bin/Debug/netcoreapp2.1/linux-arm/publish/ YourPiHostName:/some/dir/on/your/pi
```

presumably you'll need to wrap it with ssh to your WSL first (assuming you already got pri key, `ssh localhost rsync -r ....` and adjust directories) but I haven't tried it

monty124, 2019-06-12:
winscp can also keep a remote directory up to date without synching all files too. thats what i use (setup and run in the background)

PR #501 (2019-06-17): Remove System.Device.Gpio.Native shim
Fixes #345 #351 
Contributes to #324 

FYI: @buyaa-n 

cc: @safern - for the yml changes
     @ericstj - for packaging changes
     @bartonjs - for PInvoke changes, specifically TimeSpec struct.

This change will remove the use of the Native shim in System.Device.Gpio and it will have us PInvoke directly into libgpiod. By doing this, our package will become simpler and more portable. It will allow it also to run in multiple Linux runtimes. These changes have been tested and validated to work on arm32.
closed 2019-06-17 by joperezr