dotnet/iot #347: Disposing pins is not always expected

krwq, 2019-03-27:
In some scenarios disposing pins is not expected. Here is the scenario I've encountered recently:
I've worked on oLED bindings - one of the oLED pins is `reset`.
Scenario is following:
- start .NET app to draw something and close it

Here is what happens instead:
- .NET app starts, draws whatever user requested
- Dispose is called
- Reset PIN is set to Low
- oLED is reset
- .NET app finishes

current solution for this problem was not to call Dispose at all.

I think we should have a way of telling controller that we don't want any cleanup to happen for the specific pin (in my case `reset` pin)

joperezr, 2019-04-10:
Interesting scenario. I kind of feel that if we wanted to do this we might leak some files for specific set of drivers. For example, the SysFS driver would need to leave some files behind in order to persist the value. I'm not 100% sure  on this, but I believe that is the case.

shaggygi, 2019-06-19:
While I'm just speculating with the work on `I2cController`, we could eventual evolve the IGpioController to manage GpioPins (or IGpioPins) instead.  This way, you could possibly have a `ShouldDispose` property like we might be adding to `I2cDevice`.

joperezr, 2019-07-09:
I don't think we want to optimize for the scenario you specified above with the OLED displays. As for the disposing pattern, we have that covered in another issue so I will close this as a dupe as the fix will be tracked by that other one

joperezr, 2019-07-09:
duplicate of #419

closed 2019-07-09 by joperezr