dotnet/iot #348: Bmx280 SPI interface missing

EthanTeece, 2019-03-27:
Hi

I have a BME280 temperature, humidity and pressure sensor for a school project which is supported in the devices for dotnet IoT.

The readme states under connection types I2C and SPI, I know SPI is unticked but It looks like someone thought about adding SPI support but then didn't.

Does this mean it's currently being implemented? I had a look at the code and I thought it was just a case of editing the CommunicationProtocol enum in BmxBase.cs but then there's a lot of very i2c specific code in that class so I'm not sure the CommunicationProtocol enum is doing much, if anything at the moment (it seems silly to only have one option).

shaggygi, 2019-03-27:
@EthanTeece this most likely means the binding only has support for I2C for now.  You might want to rename this issue to something like "Add SPI interface to Bmx280 binding" so it will be on the radar for someone to pick it up.

EthanTeece, 2019-03-27:
@shaggygi thanks, I understand this is probably low priority so I won't get my hope up haha. I'd be happy to help test it though!

joshfree, 2019-04-02:
tagging this as an `up-for-grabs` issue so it's more easily discoverable

joperezr, 2019-04-10:
@ghost In case you want to take this yourself I believe we have a version of the SPI implementation here: https://github.com/dotnet/corefxlab/blob/8aaec86b7bb83fbb63a6077d37a23428800d2a81/samples/System.Devices.Gpio.Samples/PressureTemperatureHumiditySensor.cs  So it shouldn't be too difficult to port that into our existing binding. As a hint, look in that file for places where we do something like:
```c#
            switch (_protocol)
            {
                case ConnectionProtocol.Spi:
```
As those would be the ones you want to add to the binding.

kurtnelle, 2019-05-21:
The issue is implement the SPI version as well. How does one grab and issue?

joperezr, 2019-05-21:
Hello @kurtnelle, if you want to take this, then you can create a PR, and mention in your description `Fixes #348` and we will assign you this issue and review the PR. Thanks a lot for  your interest in helping us!

closed 2023-02-09 by krwq