dotnet/iot #355: MyGet feeds not updated

techniq, 2019-04-07:
I was looking to update the packages to pull in the fixes for [ADS1115](https://github.com/dotnet/iot/commit/ac92a6069024905b3ceb79911bbfba8c11e35d50) which was merged on 3/31 but it appears the last time the MyGet feeds were updated is 3/27.

- https://dotnet.myget.org/feed/dotnet-core/package/nuget/System.Device.Gpio
- https://dotnet.myget.org/feed/dotnet-core/package/nuget/Iot.Device.Bindings

I also saw packages on nuget.org but they are a little older than MyGet

- https://www.nuget.org/packages/System.Device.Gpio/0.1.0-prerelease.19171.3
- https://www.nuget.org/packages/Iot.Device.Bindings/0.1.0-prerelease.19171.3

krwq, 2019-04-09:
cc: @safern

joshfree, 2019-04-10:
> the last time the MyGet feeds were updated is 3/27.

I believe @tarekgh and @dagood may have been looking into this

safern, 2019-04-10:
This is because the MyGet feed is full. @mmitche has been trying to get this sorted out with MyGet people but haven't had response I believe.

dagood, 2019-04-10:
Yeah, see https://github.com/dotnet/core-eng/issues/5070 for details. The current recommendation is to use the blob feeds like https://dotnetfeed.blob.core.windows.net/dotnet-core/index.json, but I don't know if IoT is configured to publish to a blob feed yet.

safern, 2019-04-10:
cc: @joperezr

joperezr, 2019-04-11:
I'll work today on making sure that we publish to blobfeed as well as to myget. That way, even if myget publish fails, there is still a place where the community can get the packages out of.

joperezr, 2019-04-17:
Just to update this thread, our myget feeds have been having a lot of issues so we haven't been able to push for a while. Instead, we are now pushing all of our official builds to a blob feed: https://dotnetfeed.blob.core.windows.net/dotnet-iot/index.json

We will eventually stop pushing packages to myget and only use the feed instead, so I will go ahead and update our documentation in the repo to point to the blob feed instead of myget in order to consume packages.

MarkCiliaVincenti, 2019-05-15:
> Just to update this thread, our myget feeds have been having a lot of issues so we haven't been able to push for a while. Instead, we are now pushing all of our official builds to a blob feed: [...]

Can you please amend https://github.com/dotnet/iot/blob/master/README.md then?

joperezr, 2019-05-15:
Will do, in fact we updated the initial comment on this issue to say that this issue is tracking the documentation update, and that is why we haven't closed this issue yet. We will get to this soon.

techniq, 2019-06-09:
Took some trial and error, but I was finally able to install/upgarde from the blob storage using 
```
dotnet add package System.Device.Gpio -v 0.1.0-prerelease.19307.2 --source https://dotnetfeed.blob.core.windows.net/dotnet-iot/index.json 
dotnet add package Iot.Device.Bindings -v 0.1.0-prerelease.19307.2 --source https://dotnetfeed.blob.core.windows.net/dotnet-iot/index.json
```

A few notes:
- Pay attentional to the url better than I did.  I was trying to use `dotnet-core` instead of `dotnet-iot` as I missed this detail between this [comment](https://github.com/dotnet/iot/issues/355#issuecomment-481840500) and this [one](https://github.com/dotnet/iot/issues/355#issuecomment-484294130)
- Pre-release installing [still](https://github.com/NuGet/Home/issues/4699) doesn't look to be available using `dotnet add package`, but I was able to provide the explicit version which I saw after the failed command without the version
> error: Unable to find a stable package System.Device.Gpio with version 
error:   - Found 76 version(s) in https://dotnetfeed.blob.core.windows.net/dotnet-iot/index.json [ Nearest version: 0.1.0-prerelease.19307.2 ]
error:   - Found 3 version(s) in nuget.org [ Nearest version: 0.1.0-prerelease.19171.3 ]
error: Package 'System.Device.Gpio' is incompatible with 'all' frameworks in project '/Users/techniq/Documents/Development/projects/iot/Iot.Playground/Iot.Playground.csproj'.

closed 2019-07-01 by krwq