dotnet/iot #366: Can't build in VS or in Rider

daveyostcom, 2019-04-16:
I cloned [the repo](https://github.com/dotnet/iot.git), then selected the DeviceApiTester solution. Something very basic is wrong, and it won't build. 
[image]

joperezr, 2019-04-16:
Did you run `.build.cmd` first? As part of our workflow we require you to run build.cmd on the root of the repo once, so that you get all of the dependencies restored and then you should be able to open the projects in VS. Rider isn't officially supported by the repo and therefore hasn't be tested yet.

daveyostcom, 2019-04-16:
No, I didn’t.

This is the first I’ve heard of a repo that can’t be used in Visual Studio without doing something on the command line first.

Is this because of a shortcoming of VS? If so, have you reported it to them?

joperezr, 2019-04-16:
Most of the repos in the dotnet org use a shared central infrastructure. In order to use it, this infrastructure must be restored, which is what is accomplished by running the build.cmd script. This is not uncommon, as it is similar to other repos that require some sort of restore (either using npm install, or restoring some build dependencies in general) before the initial build. If you check all of the rest of our repos, corefx, coreclr, etc, they all require you to restore these tools first, before opening VS.

I'll go ahead and close this given that the question has been answered. Feel free to re-open if you still have some questions/concerns.

daveyostcom, 2019-04-16:
Thanks. Their ReadMe says I can load their stuff using nuget. I failed. I set up Rider NuGet to use their source, checked the prerelease checkbox, and couldn't find their stuff. Could you try it please?

Thanks

daveyostcom, 2019-04-16:
Here's what I tried:
```
using System;
using System.Device.Gpio; // NuGet source: https://dotnet.myget.org/feed/dotnet-core/package/nuget/System.Device.Gpio

namespace iotapp {

  class Program {
  
    static void Main(string[] args) {
      Console.WriteLine("Hello World!");
    }
    
  }
}
```

joperezr, 2019-04-16:
Reopening to track the additional questions.

So then is the issue that you are trying to consume the packages but you can't seem to be able to install the package? I thought the original issue was that you were trying to load a solution from our repo into VS. If all you want is to consume our package, you can either try using our [latest prerelease version available in NuGet.org](https://www.nuget.org/packages/System.Device.Gpio/0.1.0-prerelease.19171.3), or alternatively you could try adding this source in order to get our daily builds: https://dotnetfeed.blob.core.windows.net/dotnet-iot/index.json

daveyostcom, 2019-04-16:
Sorry. I thought I was writing that response to Rider support. (Wrong window open.) I edited my response above for this location. But nevermind.

closed 2019-04-16 by daveyostcom