dotnet/iot #380: Error adding the Nuget Package to a new console project

raffaeler, 2019-04-23:
Platform: Windows 10 x64
NetCore: 3.0.100-preview3-010431
Visual Studio 2019 16.1.0 Preview 1.0
Nuget Package version:
```
    <PackageReference Include="System.Device.Gpio" Version="0.1.0-prerelease.19171.3" />
```

When adding the Nuget Package to a new console project, Visual Studio shows the following error in the error lists:
```
Expected 1 export(s) with contract name "NuGet.PackageManagement.VisualStudio.IScriptExecutor" but found 0 after applying applicable constraints
```

This is comes from the output window:
```
Microsoft.VisualStudio.Composition.CompositionFailedException: Expected 1 export(s) with contract name "NuGet.PackageManagement.VisualStudio.IScriptExecutor" but found 0 after applying applicable constraints.
   at Microsoft.VisualStudio.Composition.ExportProvider.GetExports(ImportDefinition importDefinition)
   at Microsoft.VisualStudio.Composition.ExportProvider.GetExports[T,TMetadataView](String contractName, ImportCardinality cardinality)
   at Microsoft.VisualStudio.Composition.ExportProvider.GetExport[T,TMetadataView](String contractName)
   at Microsoft.VisualStudio.Composition.ExportProvider.GetExport[T](String contractName)
   at Microsoft.VisualStudio.Composition.ExportProvider.GetExport[T]()
   at Microsoft.VisualStudio.Composition.ExportProvider.GetExportedValue[T]()
   at Microsoft.VisualStudio.ComponentModelHost.ComponentModel.GetService[T]()
   at NuGet.PackageManagement.VisualStudio.GlobalProjectServiceProvider.GetGlobalService[T]()
   at NuGet.PackageManagement.VisualStudio.VsProjectScriptHostService.<>c__DisplayClass2_0.<.ctor>b__0()
   at System.Lazy`1.CreateValue()
   at System.Lazy`1.LazyInitValue()
   at System.Lazy`1.get_Value()
   at NuGet.PackageManagement.VisualStudio.VsProjectScriptHostService.<ExecutePackageInitScriptAsync>d__4.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at NuGet.PackageManagement.BuildIntegratedRestoreUtility.<ExecuteInitPs1ScriptsAsync>d__0.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd(Task task)
   at NuGet.PackageManagement.NuGetPackageManager.<ExecuteBuildIntegratedProjectActionsAsync>d__78.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd(Task task)
   at NuGet.PackageManagement.NuGetPackageManager.<ExecuteNuGetProjectActionsAsync>d__76.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at NuGet.PackageManagement.NuGetPackageManager.<ExecuteNuGetProjectActionsAsync>d__75.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd(Task task)
   at NuGet.PackageManagement.NuGetPackageManager.<ExecuteNuGetProjectActionsAsync>d__74.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at NuGet.PackageManagement.UI.UIActionEngine.<ExecuteActionsAsync>d__17.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd(Task task)
   at NuGet.PackageManagement.UI.UIActionEngine.<>c__DisplayClass12_0.<<PerformActionImplAsync>b__0>d.MoveNext()
```

Rebuilding and running the project works as expected.

joperezr, 2019-04-23:
I don't believe this is a problem with our package, but with VSs package manager instead, so I'm closing this for now. If you have a consistent repro for this on new projects, please reopen this a let us know which version of VS you are using so that we can investigate further.

closed 2019-04-23 by joperezr

raffaeler, 2019-04-23:
Thank you I just found this to be a Visual Studio bug.