dotnet/iot #392: Regarding device binding guidelines; async or synchronous APIs, properties or methods

jimutt, 2019-04-27:
Hi, I'm very much appreciating all work being put into the IoT libraries. I'm having a look round the repo to try get a wider picture of what's going on. This question is regarding device bindings, and how to achieve a consistent and predictable experience for the library consumers.

When it comes to reading sensor values the current [conventions](https://github.com/dotnet/iot/blob/master/Documentation/Devices-conventions.md) states "Prefer property (i.e. Temperature) rather than method (i.e. GetTemperature)". I guess that the reasoning behind that recommendation is to provide an abstraction that allows for simplest possible hardware access, which it does. But in some cases I think might be contributing to some potentially problematic implementations. 

If you look at the [DHT sensor bindings](https://github.com/dotnet/iot/tree/master/src/devices/Dhtxx) the public API for reading temperature values is designed according to the guidelines:
```C#
using (DhtSensor dht = new DhtSensor(device))
{
    Temperature temperature = dht.Temperature;
    double humidity = dht.Humidity;
}
```
Though the implementation is doing synchronous 1-wire bit-banging. Which means that the get accessor for the `Temperature` property will take a pretty long time to complete as it needs to initialize communication and read 40 bits of data. 

I think it's fair to say that the majority of .NET developers does not consider reading a property as something that could be potentially very time consuming. I also believe that consistency and predictability is good. But with the current implementation in the mentioned library it will only read the sensor if it's more than 1 second since the last read (which is good and according to manufacturer recommendations). So sometimes it will return immediately but sometimes with a large delay. I personally do not think that this is not good API design as it gives very unreliable timing results if the user is not very well aware of the internal implementation. 

I think the current design guidelines needs to be discussed and I'd personally prefer if very time consuming tasks like reading one wire devices were not done from property accessors, as I'm afraid it will lead to many mistakes by future users not very familiar with the library implementations and the specific hardware they're trying to interface with. 

I wouldn't consider this a big problem if this wasn't a .NET Foundation provided library of device bindings. But as that's the case I think a high level of consistency and reliability needs to be maintained. Though I'm fully aware that the IoT libraries are still in preview, I just want to make sure that this is being payed attention to.

joperezr, 2019-04-29:
Hello @jimutt thanks a lot for your input. I'm tagging @krwq as he has been looking into this topic for a while. I do agree that your logic about a property is usually understood as a simpler/low-time-consuming API, so it is worth to disucssing if having a GetTemperature() would make more sense in this case.

> I wouldn't consider this a big problem if this wasn't a .NET Foundation provided library of device bindings.

A small comment around this. Even though the System.Device.Gpio library is part of the .NET Foundation and have the same level of support than the rest of our libraries, the device bindings library is not under the same support since this is more community-driven.

krwq, 2019-04-29:
This is good feedback.
As @joperezr has mentioned the devices have softer rules and we are still figuring out the guidelines for this repository so take any recommendations with the grain of salt.

I'm thinking of two solutions here:
- Temperature sensor in cases like this should keep on reading in a background and property always returns immediately
  - if user asks for thermometer they probably will continuously read the temperature anyway so IMO it makes sense to do so - if user does not want to read the temperature they should dispose the thermometer - other option is to make reading process more open: allow some configuration or expose some kind of ReadAsync which would do the actual reading while Temperature would just report recently read value (in case of first call without calling Read it will throw)
- GetTemperatureAsync for slow implementations and just Temperature for fast

I think we should update the guidelines to account for cases like this.

@tarekgh @JeremyKuhne - any opinions?

I think until we have official guidelines any of the above is fine.

jimutt, 2019-04-30:
@joperezr @krwq  Thanks for explaining the situation for the device bindings library and taking the time to go through my input.

I've got to admit that I'm mainly used to working with embedded systems programming in the range from assembly to the level of abstraction that Arduino-like libraries offers. So I probably don't have very good ideas about which way is best to go in this case. But I think all of your three suggestions makes sense. 

Personally I'd most likely prefer "make reading process more open: allow some configuration or expose some kind of ReadAsync" so that I can read values from the sensor exactly when needed. But from a general user perspective I believe managing the reading operation in the background for temperature sensors (and similar) makes sense and it would probably be the most convenient way to handle it for most users. 

Though doing too much stuff in the background in multiple places could potentially also be problematic if it's the kind of time sensitive bit banging used in the mentioned case. But to be fair you don't really want much of too much of that on a non-realtime OS anyway. And I see that native 1w support for at least Linux devices seems to be on the way. 

I believe alternative 2 (the more manual approach) is the one that that reduces the risk of potential problems the most. But I'm sure you'll be able to decide what fits best for the context of these libraries!

Ellerbach, 2021-07-08:
[Triage] After quite some time and quite some efforts to harmonize how to best do it, we came with the following design which should be implemented almost everywhere now (still few bindings need some rework):

* We do have properties for instantaneous measurements or configuration. We try not use them if the value is always changing.
* We do have a `TryGetSomething` when like in the case of the DHT, there are chances that the read value won't be correct.
* We do have `GetSomthing` when reading is expensive and taking some time.
* Few bindings do have Async variants but it's not the general case, as most of the time, we've seen that you really wait for the value you want to get, it's not a fire and forget and let's see when the result will come. That's most likely a specificity of embedded devices.

So closing this discussion and thanks again for the feedback as it helped design what we have today.

closed 2021-07-08 by Ellerbach