dotnet/iot #399: Humidity and CompensateHumidity readings for BME280

MarkCiliaVincenti, 2019-05-03:
For BME280, you can read the pressure and the temperature (including compensate values) but so far there's nothing for humidity. Any chance for this to be implemented please?

joperezr, 2019-05-06:
@MarkCiliaVincenti Adding this shouldn't be too hard, any chance you want to take a stab at it 😄 ? We are happy to guide you and help you with any questions that you may have.

MarkCiliaVincenti, 2019-05-14:
> @MarkCiliaVincenti Adding this shouldn't be too hard, any chance you want to take a stab at it 😄 ? We are happy to guide you and help you with any questions that you may have.

It's not as easy as one would think. There's a lot of stuff added onto BME280 over BMP280 so some architectural changes are needed to accommodate them, not just a case of adding a couple of methods to the Bme280 class.

joperezr, 2019-05-15:
mm I'm not sure I follow. If this functionality is strictly only for Bme280 class, then it should just be adding a couple of methods to that file, you shouldn't need to touch BMP280 as that class won't have this functionality.

MarkCiliaVincenti, 2019-05-15:
> mm I'm not sure I follow. If this functionality is strictly only for Bme280 class, then it should just be adding a couple of methods to that file, you shouldn't need to touch BMP280 as that class won't have this functionality.

It's not just a case of adding methods. BME280 requires a vaster CalibrationData.cs and a vaster Register.cs. It depends on what architecture path you want to take. Do you completely separate the two or do you do base classes for CalibrationData and Register classes too?

joperezr, 2019-05-15:
Oh I see now, that makes sense. It would take a bit of refactoring I guess to have that work, since you would have to set the calibration data on each individual class. The register classes don't need to change IMO, since it's fine if one of the registers in the enum is not used by one of the classes.

kurtnelle, 2019-05-20:
You can borrow code from my project https://github.com/kurtnelle/BME280Logger or monoculture's project to get the humidity logic.

PR #435 (2019-06-19): BME280 - Humidity Support
This pr adds support for humidity for BME280 devices.

- Moved the responsibility of reading calibration data to the implemented devices, away from `CalibrationData`.
- Added additional registers.
- Added method to read humidity data and to apply calibrations.

This resolves #399.
closed 2019-06-19 by joperezr