dotnet/iot #416: Examples don't work

katzz0, 2019-05-11:
I have DHT22 sensors which works normally. I checked that by the default python script from Adafriut. I made a decision to give .net core on rpi platform a try with a simple example from the examples folder. I used Dhtxx example but it doesn't work.

This is my code:
```
using System;
using System.Threading;
using Iot.Device.DHTxx;

class Program
{
    static void Main(string[] args)
    {
        Console.WriteLine("Hello DHT!");

        using (DhtSensor dht = new DhtSensor(26, DhtType.Dht22))
        {
            while (true)
            {
                Console.WriteLine($"Temperature: {dht.Temperature.Celsius} °C, Humidity: {dht.Humidity} %");

                Thread.Sleep(2000);
            }
        }
    }
}
```

Project file:
```
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>netcoreapp2.2</TargetFramework>
  </PropertyGroup>

  <ItemGroup>
    <PackageReference Include="IoT.Device.Bindings" Version="0.1.0-prerelease.19251.2" />
    <PackageReference Include="System.Device.Gpio" Version="0.1.0-prerelease.19251.2" />
  </ItemGroup>

</Project>
```
I wasn't able to install System.Device.Gpio and IoT.Device.Bindings by using commands from the main readme.md so I just used the nearest proposed version by dotnet add.

Do I do something wrong or the project isn't ready yet for real usage (only people with deep .net knowledge can use it)?

joperezr, 2019-05-14:
Hey @katzz0 thanks a lot for your feedback. I assume that since you have some code running with python that we can disregard any problems related to bad connections. Just to double check, do you have the right pin specified? From your code above, you should be using pin 26 in the BCM numbering, not the physical layout. 

> Do I do something wrong or the project isn't ready yet for real usage (only people with deep .net knowledge can use it)?

From what I can tell you didn't do anything wrong, unless the gpio pin that you specified is incorrect. As for your other question, this library is definitely ready for real usage. We build two different components out of this repo, 1) the main library (System.Device.Gpio) which contains all the protocols in order to talk to individual sensors and devices, and 2) A collection of classes that wrap different sensors (we call these: device bindings) and expose an easy way to talking to them in order to get data from them. These classes internally will call System.Device.Gpio library. Most of these device bindings have been ported by the community, so we don't yet have a good story on testing on the actual hardware for some of them so we have been reviewing based on the data sheets for the components. That is the case for the DhtSensor. Are you getting any exception after the NaN or do you get any additional information when attached to the debugger?

MarkCiliaVincenti, 2019-05-15:
I had the same issue. Use the TryGetTemperatureAndHumidity instead rather than what is suggested in the samples.

joperezr, 2019-05-15:
@katzz0 if what @MarkCiliaVincenti suggests works, we should log an issue to fix the methods that are not working properly, and at least in the meantime switch the examples to use the API that does work. 

cc @ZhangGaoxing and @Ellerbach who added this binding.

MarkCiliaVincenti, 2019-05-15:
Here's my code:

`                using (DHTSensor inside = new DHTSensor(InsideGpioPin, DhtType.Dht22))
                {
                    if (!inside.TryGetTemperatureAndHumidity(out Temperature insideTemp, out insideHumidity))
                    {
                        //Console.WriteLine(" failed obtaining inside readings");
                        return;
                    }

                    insideTemperature = insideTemp.Celsius + InsideTemperatureOffset;
                    insideHumidity += InsideHumidityOffset;
                }`

MarkCiliaVincenti, 2019-05-15:
OK I can confirm. My code works with the old (MyGet version) package.
Couldn't get the new (dotnetfeed blob version) package to read from DHT22 either. The TryGetTemperatureAndHumidity method doesn't exist anymore and IsLastReadSuccessful is always false.

joperezr, 2019-05-15:
mm I see, interesting, I'll take a look at this and see what changed.

ZhangGaoxing, 2019-05-16:
The last change was made by me. I don't have DHT22, I just make sure that the code I committed works correctly on my DHT11 and DHT12. Interestingly, the old code always reads errors on my DHT11 (just like the two above), so when I added DHT12, I changed the original code.

According to the datasheet, the DHT series of sensors are read in the same way, which is not a problem. The code I committed was also written entirely in accordance with the datasheet. Reading errors must be timing errors. I've made several key parts loop and stopped the reading as soon as it went out of time, but krwq prevented me from doing so(https://github.com/dotnet/iot/pull/318#discussion_r266714856).

Another possible reason is the version of the sensor. There are some fake DHT sensors in the market.

Give me some time and I will try to solve this problem. I don't have any equipment on hand now. I'll start processing this weekend.

katzz0, 2019-05-16:
@joperezr thanks for the detailed explanation! I see that this behaviour is a combination of a few smaller problems. I just expected to run the sample as people usually with new things to try it out and it seems I picked up the wrong sensor to start with))) I checked the BMP280 binding and it works normally through I2C but I needed to change the default address since it was 0x77 but mine was 0x76. I think it would be nice to have the collection of addresses predefined since the majority of sensors have an ability to switch to an alternative address.

Regarding debugging, I didn't try debugging since I had problems with `dotnet run` and there is an issue for that https://github.com/dotnet/iot/issues/345. But now I know how to handle it via `dotnet publish -r linux-arm`.

I have both sensors DHT11 and DHT22 (it's actually 3-wire bigger module to be mounted on a surface). I can confirm the python scripts work and values are read correctly from that sensors. I could try to play with that to figure out the problem but next two weeks I'll be out of home and won't be able to take my rpi with me. But I can test both sensors after two weeks.

That might be helpful - I found that code in the internet https://broersa.github.io/dotnetcore/2017/12/29/raspberry-pi-thermometer-using-raspbian-dotnetcore-and-dht22-sensor.html and it also doesn't work.

@ZhangGaoxing I also think that's a timing issue. I have Raspberry Pi 3 Model B but it can work differently on Rpi 2 Model B and Rpi 3 Model B+. Finally 6 combinations should be tested to be sure that DHT11 and DHT22 work on Rpi.

RobinTTY, 2019-05-16:
I was able to reproduce this problem using a Dht11 both on Raspberry Pi 3 Model B and Model B+. I also tested on both Linux and Windows with the same result. I remember timing also being an issue with 1-wire on UWP.

Ellerbach, 2019-05-16:
> @ZhangGaoxing I also think that's a timing issue. I have Raspberry Pi 3 Model B but it can work differently on Rpi 2 Model B and Rpi 3 Model B+. Finally 6 combinations should be tested to be sure that DHT11 and DHT22 work on Rpi.

Yes, the timing is very sensitive. As explained in the readme, this can break depending on the platform you are using. I'm far from home where I have everything to test. So won't be able to best at least for a week. My code worked with DHT11 and DHT22, I have code running for quite some time wihtout any issue in .NE Core deployed thru Azure IoT Edge but I haven't updated it with the latest .NET Core IoT bits. 
So in short, will test it as soon as I can.

MarkCiliaVincenti, 2019-05-16:
> Yes, the timing is very sensitive. As explained in the readme, this can break depending on the platform you are using. I'm far from home where I have everything to test. So won't be able to best at [...]

Not sure if this is relevant, but I tested on .NET Core on Raspbian on Raspberry Pi 3 Model B+ with DHT22 from dx.com that cost me some 4 euro, so they're probably fake :)

Rickvanderwaal, 2019-05-28:
I am experiencing the same issues as described above.

Running on Raspbian on Raspberry Pi 3 Model B+ with DHT22 (AM2302).
Package Iot.Device.Bindings version 0.1.0-prerelease.19278.2.

Testing with the python library from adafruit was successful.
```
import Adafruit_DHT
sensor = Adafruit_DHT.AM2302
pin = 26 
humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)
print 'Temperature: {0:0.1f}*C'.format(temperature)
print 'Humidity: {0:0.1f}%'.format(humidity)
```
It would be nice if i could read the sensor in .NET Core. I am using the following code:
```
static void Main()
{
	using (DhtSensor dht = new DhtSensor(26, DhtType.Dht22))
	{
		while (true)
		{
			Console.WriteLine($"Success {dht.IsLastReadSuccessful}, Temperature: {dht.Temperature.Celsius.ToString("0.0")} °C, Humidity: { dht.Humidity.ToString("0.0")} % ");
			Thread.Sleep(1000);
		}
	}
}
```

Above example gives me the following output:
[image]

None of these values are correct.
Am i doing something wrong?

joperezr, 2019-05-28:
cc @ZhangGaoxing do you mind taking a look to see if that's expected?

ZhangGaoxing, 2019-05-29:
OK, I'll check it again.

Rickvanderwaal, 2019-06-02:
I looked into the source code and found the following issue,

Reading the bytes from the DHT22 seems to be correct but translating the bytes to humidity/temperature seems to be the issue.

DhtSensor.cs line 297 results in a temperature of 1.3°.
```
_readBuff[2] + (_readBuff[3] & 0x7F) * 0.1
```

When i replace the formula with the following code, the result of the temperature is 25.9°. Which is correct.
```
(_readBuff[2] << 8 | _readBuff[3]) * 0.1
```

The same issue with humidity.
DhtSensor.cs line 305 results in a humidity of 9.9%.
```
_readBuff[0] + _readBuff[1] * 0.1
```

The following code gives me a humidity of 59.1% which seems correct.
```
(_readBuff[0] << 8 | _readBuff[1]) * 0.1
```

katzz0, 2019-06-08:
My code still doesn't work. I tried DHT11 and DHT22 and reading isn't successful so the result is the same:

```
Hello DHT!
Success False, Temperature: 179769313486232000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.0 °C, Humidity: 179769313486232000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.0 %
Success False, Temperature: 179769313486232000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.0 °C, Humidity: 179769313486232000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.0 %
Success False, Temperature: 179769313486232000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.0 °C, Humidity: 179769313486232000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.0 %
Success False, Temperature: 179769313486232000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.0 °C, Humidity: 179769313486232000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.0 %
Success False, Temperature: 179769313486232000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.0 °C, Humidity: 179769313486232000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.0 %
```

These are my files:
Program.cs
```
using System;
using System.Threading;
using Iot.Device.DHTxx;

namespace dht22
{
    class Program
    {
        static void Main()
        {
            Console.WriteLine("Hello DHT!");

            using (DHTSensor dht = new DHTSensor(26, DhtType.Dht22))
            {
                while (true)
                {
                    Console.WriteLine($"Success {dht.IsLastReadSuccessful}, Temperature: {dht.Temperature.Celsius.ToString("0.0")} °C, Humidity: {dht.Humidity.ToString("0.0")} %");

                    Thread.Sleep(2000);
                }
            }
        }
    }
}
```

dht22.csproj
```
<Project Sdk="Microsoft.NET.Sdk">

  <PropertyGroup>
    <OutputType>Exe</OutputType>
    <TargetFramework>netcoreapp2.2</TargetFramework>
  </PropertyGroup>

  <ItemGroup>
    <PackageReference Include="IoT.Device.Bindings" Version="0.1.0-*" />
    <PackageReference Include="System.Device.Gpio" Version="0.1.0-*" />
  </ItemGroup>

</Project>
```

I use `dotnet publish -r linux-arm --self-contained false` to create the deployable program.

katzz0, 2019-06-08:
When I tried  to clone Iot project through git and build the sample for DhtXX I got it working but it asked me to install .Net Core 3.0 preview and produced the build for 2.1. Below you can see the result:

[image]

IMHO it doesn't seem stable. And I'm keen on making the code in my previous message working.

Rickvanderwaal, 2019-06-08:
@katzz0 Which specific version of IoT.Device.Bindings are you using because you are using a wildcard in the .csproj. Have you added the dotnet-iot package source for nuget ? (https://dotnetfeed.blob.core.windows.net/dotnet-iot/index.json)

Also try to publish your project in release mode because of better performance. `dotnet publish --configuration release --runtime linux-arm`

You said it worked with the latest version of the iot project. Does it work for both devices dht11 & dht22?

benbuckland, 2019-06-12:
Hi, i'm facing the same issue, returns data sporadically but even when it does it appears to be wrong :( Im using 19310.5 version of packages.

joperezr, 2019-06-12:
The version that @benbuckland mentions is Preview6 which was released yesterday, so this does mean that the problem repros on latest build (so I'm reopening the issue). There is a PR out by @Rickvanderwaal that will hopefully fix this issue, and should be merged pretty soon.

benbuckland, 2019-06-12:
@joperezr Very happy to help out so please let me know if there is anything I can do!

joperezr, 2019-06-12:
Once we merge that PR, it would be great if you can update to reference that version of the package to see if that fixes the issue for you.

Rickvanderwaal, 2019-06-13:
@benbuckland you are right, i only solved the incorrect measurements for dht22 because i think that is the most important.
I agree we should work om the reliability. I do have Some idea’s but those are not backwards compatible. I think the current public methods/properties of the DHTSensor class are confusing and maybe that is the problem for reliability too.
Maybe we should create another issue for working on reliability?

joperezr, 2019-06-14:
> Maybe we should create another issue for working on reliability?

Agreed. Let's use this one to make sure that the actual device works (although not reliable) which should be fixed with @Rickvanderwaal PR. We can then log an additional one just for reliability.

closed 2019-06-17 by joperezr

MarkCiliaVincenti, 2019-09-07:
> OK I can confirm. My code works with 19171.3
> Couldn't get the latest prerelease packages to read from DHT22 either. The TryGetTemperatureAndHumidity method doesn't exist anymore and IsLastReadSuccessful is always false.

A number of months have passed so I thought I'd give it a go seeing that this issue was closed. The result is the same.

Old code that works (19171.3):
`using (DHTSensor inside = new DHTSensor(InsideGpioPin, DhtType.Dht22))
                {
                    if (!inside.TryGetTemperatureAndHumidity(out Temperature insideTemp, out insideHumidity) || inside.Temperature.Celsius > 60d || inside.Humidity > 100d)
                    {
                        //Console.WriteLine(" failed obtaining inside readings");
                        return;
                    }

                    insideTemperature = insideTemp.Celsius + InsideTemperatureOffset;
                    insideHumidity += InsideHumidityOffset;
                }`

New code that doesn't work (IsLastReadSuccessful is permanently false), 19456.1:
`using (var inside = new Dht22(InsideGpioPin))
                {
                    if (!inside.IsLastReadSuccessful)
                    {
                        Console.WriteLine(" failed obtaining inside readings");
                        return;
                    }

                    insideTemperature = inside.Temperature.Celsius + InsideTemperatureOffset;
                    insideHumidity = inside.Humidity + InsideHumidityOffset;
                }`

benbuckland, 2019-09-07:
@MarkCiliaVincenti You need to call .Temperature or .Humidity first and then check IsLastReadSuccessful. I am successful about 1 in 5 times of reading the sensor, have upgraded to the RPI4 thinking that the increased processing power would help the super time sensitive reading of the sensor but alas no luck.

MarkCiliaVincenti, 2019-09-07:
Thanks, I'll try that later. Strange regarding the reliability. I estimate
I was getting successful readings about 11 out of 12 times from the old
build.

MarkCiliaVincenti, 2019-09-08:
I confirm. Readings are available but the reliability is considerably poorer than with the NuGet package version 19171.3. Reverting back to that version.

benbuckland, 2019-09-08:
Will take a look and compare commits, see if I can identify the problem.

benbuckland, 2019-09-08:
I have taken a look and cant see any change that would affect the reliability of reading. I'm going to roll back to 19171.3 and see if reliability increases for me also.

krwq, 2019-09-08:
Since this is 1-wire I wouldn't expect reliability but I bet we can bet slightly better results than 10%. I know linux has a way of enabling 1 wire through kernel driver which might work better here

MarkCiliaVincenti, 2019-09-09:
It's not slightly better. It's massively better in the old build. Also in a
Python library I used before it was very rare that a reading was missed.

PR #430 (2019-05-21): Try to fix DHT 1-wire read error
try to fix #416 

try this @katzz0 @MarkCiliaVincenti

It's OK for me.
[image]
PR #469 (2019-06-17): Resolved issue #416 - translating the dht22 bytes to temperature
Fixes #416