dotnet/iot #418: Azure DTDL & IoT Plug and Plug Support

shaggygi, 2019-05-12:
There were recent announcements regarding Azure's new Digital Twin Definition Language (DTDL) and [IoT Plug and Play](https://github.com/Azure/IoTPlugandPlay) at Build 2019.  We should review and collaborate with that team as this repo and APIs could "plug and play" nicely with each other.

There are scenarios that can help with System.Device.* tooling and code generation for device bindings.  This would also be extremely helpful to better define bindings as this has been a huge gap/struggle trying to enforce consistent standards between contributors.  Device GPIO pin assignments, I2C/SPI settings, and register definitions are just a few examples where it would benefit.  In addition, I could also see these types of DTDL docs being used for [external configuration in containers](https://github.com/dotnet/iot/issues/8).

I already created a few issues on the [IoTPlugAndPlay repo](https://github.com/Azure/IoTPlugandPlay/issues) as I see some limitations related to DTDL.  For example, it doesn't currently [support unsigned data types](https://github.com/Azure/IoTPlugandPlay/issues/8).

It might be worth doing an exercise of defining a few or our current bindings using DTDL and see what issues/ideas/innovation can be derived.

Thoughts?

//cc @joperezr @joshfree  @richlander

joperezr, 2019-05-15:
Thanks for logging this @shaggygi. In fact we have worked closely with the Azure IoT team and I believe that this is definitely something we want to work closely with them. We can use this issue for planning what the work on our side would look like, and also to gather the ideas of what we would need from the Azure team's side.

shaggygi, 2019-05-15:
@shaggygi Sounds good.  Until then, I'll keep adding issues to their repo on thoughts.  I might create a temp repo under my name to play around with some formats.  If so, I'll post link here for you to chime in.  I'd rather do that to not clutter this repo.

Also, if you get wind of [early bits](https://github.com/Azure/IoTPlugandPlay/issues/6)... send them my way so I can tinker around with them.  Interesting stuff 🎉

shaggygi, 2019-06-10:
Just a side note... if Microsoft wants to get serious about this topic, it should try to put some weight behind the support for [Common Data Model (CDM)](https://github.com/microsoft/CDM) and [Open Data Initiative (ODI)](https://www.microsoft.com/en-us/open-data-initiative).  It seems like a better bigger picture and appears like the [IoT Plug and Play DTDL](https://github.com/Azure/IoTPlugandPlay) should be based on CDM format instead.