dotnet/iot #423: Addition of binding for SSD1351

Frankenslag, 2019-05-15:
Hallo, I would like to propose adding bindings for the SSD1351 OLED driver device. This is a colour OLED driver and there is a breakout board available at 

[Colour 1.27"](https://www.adafruit.com/product/1431)
[Colour 1.5"](https://www.adafruit.com/product/1673)

[Datasheet](https://cdn-shop.adafruit.com/datasheets/SSD1351-Revision+1.3.pdf)

My intention would be to follow the general way that the ssd1306 was produced.

I would be happy to take this on but please be gentle with me as this will be my first time contributing.

joperezr, 2019-05-15:
Hey @Frankenslag we are glad to have you being part of this community! Having this binding would be great so we will gladly review your (future) PR. We have some docs around [how to contribute a binding](https://github.com/dotnet/iot/tree/master/src/devices#contributing-a-binding) which have all sorts of good information that should help you get started, specifically I would advice on following the steps [here](https://github.com/dotnet/iot/blob/master/tools/templates/DeviceBindingTemplate/README.md) that show how to create a brand new binding project using a template that was contributed by @shaggygi. That said, if you have any questions around the documentation or the process, please feel free to ping us in this thread.

Frankenslag, 2019-05-15:
O.K. I am happy for you to assign it to me. (or do I have to grab it in some way ?)

closed 2019-10-03 by Frankenslag