dotnet/iot #433: No example for PWM

djjaysmith, 2019-05-17:
There are no examples for using PWM.

Therefore, it is difficult for a noobie like myself to use the library.

Example code should always be full and complete and aimed at those with little or no knowledge.

shaggygi, 2019-05-17:
@djjaysmith Sorry about that as we do need this.  I plan to kick off a new binding soon that includes PWM support (#225).  However, it will take some time as I want to focus on adding feedback to repo where this type needs improvements (interfaces and such).

djjaysmith, 2019-05-17:
Is it not at least possible to post an example here, to enable me to get started?

I have the following, but am getting an exception and am basically stuck!

  PwmController pwmController = new PwmController();
 pwmController.OpenChannel(16, 16);

"The chip number 16 is invalid or is not enabled."

shaggygi, 2019-05-17:
I honestly have not used the related APIs yet.  We do have some basic commands you can try out using the [DeviceApiTester utility](https://github.com/dotnet/iot/tree/master/tools/DevicesApiTester).  Hope this helps for now.

joperezr, 2019-05-20:
Hello @djjaysmith thanks a lot for your submission. Seems like what you are trying to do is correct except that I believe that the channel you have seems to be invalid. It is worth noting that today our main library (the PWMController object you are using) only supports hardware PWM, which is based on a pwm channel and not on a Gpio pin. Do note that the pwm channel is **DIFFERENT** than the pin  number. One PWM Channel may be associated to one or many Pins, which is why we use different numbering and naming when talking about pwm. Are you using a raspberry pi? If so, then I would expect you should be using only either channel 0 or 1 which are the only two supported by the Raspberry Pi. Here is a pinout that shows this:

[image]

Basically as you can see, using pin BCM18 would be using pwm channel 0, so you should be able to control that pin using hardware pwm with our class. We will be adding software PWM support on a pin in the future as well, we just haven't got to it yet. If you want to double check which channels are open in your device, try searching which folders do you see under `/sys/class/pwm` since that is the interface that we use for hardware PWM, you should see a folder in there for each channel that is open, for example, a /sys/class/pwm/pwm0 folder to control PWM Channel 0.

TjWheeler, 2019-07-04:
I've put up a post on this as I got it all working with a 12v PWM Fan.  So for an example see my [post](http://blog.timwheeler.io/building-a-pwm-fan-controller-with-dotnet-iot/).  I've also put up some code on this [here](https://github.com/TjWheeler/dotnet-iot-demos).

joperezr, 2019-07-05:
@TjWheeler awesome, thanks for sharing!

krwq, 2019-07-05:
@TjWheeler for fan it actually might be useful to directly use DCMotor class, it will abstract away using PWM for you (although does not support tachometer input for better speed control yet)

shaggygi, 2019-07-21:
It appears there are a few samples sprinkled throughout bindings.  Use the following as examples.  @joperezr We could probably close this now.

https://github.com/dotnet/iot/tree/master/src/devices/DCMotor/samples
https://github.com/dotnet/iot/tree/master/src/devices/SoftPwm/samples
https://github.com/dotnet/iot/tree/master/src/devices/Servo/samples
https://github.com/dotnet/iot/blob/master/src/devices/Buzzer/Buzzer.cs

krwq, 2019-07-22:
We should add references to that somewhere in our docs.

Perhaps we should have similar README.md files as for bindings somewhere for System.Device stuff

shaggygi, 2019-07-22:
I'm thinking we could point out a few bindings that use related types when creating the sections for #100 or in the [Interface Knowledge Base](https://github.com/dotnet/iot/tree/master/Documentation#interface-knowledge-base)

Ellerbach, 2020-12-22:
And documentation added here for RPI on how to enable the hardware PWM: https://github.com/dotnet/iot/blob/master/Documentation/raspi-pwm.md
So closing this issue as I feel it's resolved

closed 2020-12-22 by Ellerbach