dotnet/iot #444: Issue with GpioController.Write

MarkCiliaVincenti, 2019-05-22:
I have the following code:

GpioController gpioController = new GpioController(PinNumberingScheme.Logical, new RaspberryPi3Driver());
gpioController.OpenPin(0, PinMode.Output);
gpioController.OpenPin(1, PinMode.Output);

GpioCommand("write 0 1");

        private static string GpioCommand(string arguments)
        {
            Process proc = new Process
            {
                StartInfo = new ProcessStartInfo
                {
                    FileName = "gpio",
                    Arguments = arguments,
                    UseShellExecute = false,
                    RedirectStandardOutput = true,
                    CreateNoWindow = true,
                    WindowStyle = ProcessWindowStyle.Hidden
                }
            };

            proc.Start();
            string output = string.Empty;
            while (!proc.StandardOutput.EndOfStream)
            {
                output += proc.StandardOutput.ReadLine();
            }
            return output;
        }

This works fine and the relay on my RPi3B+ is switched off.

However if I try this instead, nothing happens:

GpioController gpioController = new GpioController(PinNumberingScheme.Logical, new RaspberryPi3Driver());
gpioController.OpenPin(0, PinMode.Output);
gpioController.OpenPin(1, PinMode.Output);
gpioController.Write(0, PinValue.High);

joperezr, 2019-05-22:
May be a dumb question, but are you sure that you want to be using the Logical Numbering Scheme when calling pins 0 and 1? those are very special pins reserved for ID_SD and ID_SC so it is usually not a great idea to use them. Also, in your first example you are not changing the modes of pin 1, which I'm not sure if that has anything to do with what you are seeing.

shaggygi, 2019-05-22:
@MarkCiliaVincenti While I didn't try your exact code, you can use the [DeviceApiTest utility](https://github.com/dotnet/iot/tree/master/tools/DevicesApiTester) and it appears to work on my RPi3.  Try the following and see if you get similar results

```console
./DeviceApiTester  gpio-write-pin -d RPi3 -p 0 -v 0
```

shaggygi, 2019-06-02:
I'd say go ahead and close this issue for no activity.  The code in the command mentioned above is similar to what is reported and appears to work.

```csharp
// Code within command.
using (GpioController controller = CreateGpioController())
{
    controller.OpenPin(Pin, PinMode.Output);
    controller.Write(Pin, Value);

    // Used to check state.
    Console.WriteLine("Press any key to exit.");
    Console.ReadKey();
}
```

MarkCiliaVincenti, 2019-06-02:
If you please allow me more time I will test this soon. I didn't find the time just yet.

shaggygi, 2019-07-21:
I consider this inactive and provided code appears to work.  @joperezr I would close and can reopen if issue arises.

closed 2019-07-22 by krwq

MarkCiliaVincenti, 2019-10-07:
I had tried this again a while back and it still didn't work for me.

MarkCiliaVincenti, 2019-10-07:
I had tried this again a while back and it still didn't work for me, @krwq.