dotnet/iot #457: Target .NET Core 3 and Standard 2.1

shaggygi, 2019-05-26:
Is it possible to go ahead and target projects to .NET Core 3.0 and Standad 2.1?  Seems like it would be okay since this is still in preview and really for release around .NET Core 3.

Would be nice to also setup to use latest C# 8 so we can begin using features, as well.  Enables us to provide feedback to other teams if we come across issues.

joperezr, 2019-05-28:
This is a two-part question so here are my thoughts around that.

With respect to the main library and device bindings targeting .NET Core 3.0 and Standard 2.1, I would rather not do that. We really want to try to keep our main library targeting .NET Standard 2.0, as we want to be able to run in all the places .NET Standard runs on, for example, .NET Core App 2.0 apps, UWP apps, .NET Framework apps, and even in some devices that only support Mono like the Raspberry Pi Zero. If we re target to .NET Standard 2.1, then we would only be able to run on .NET Core 3.0 apps and maybe in Xamarin's new version of the framework, so we really don't want to do this unless we really depend on a new API that is only present in .NET standard 2.1, and even then, I would argue that we should dual-target (meaning we should both target .netstandard 2.0 and 2.1) so that we don't drop support. Same goes for the bindings targeting 2.1 instead of 3.0. We want our bindings to be compatible with as much apps as possible, and having them target .NET Core 2.1 allows us to reach more apps that if we limit it to just .NET Core 3.0+.

And with respect to C#8 I totally agree, we should definitely be dogfooding the new C#8 features in our projects so this we should do wherever possible.

shaggygi, 2019-05-28:
Response on .NET Core 3 is understood and makes more sense.  I mainly was asking as I thought I read the new default interface feature in C#8 required a later runtime.  I don't think it is a requirement on something I was tinkering with, so I can hold off on that.

Regarding C# 8 preview, what do we need to turn on within local projects, repo files?  Anything need to be changed in build pipeline to work, as well?

Thanks

joperezr, 2019-05-28:
I think it depends on which feature but I don't believe that all of them require the latest runtime, but I may be wrong. I'll dig more info on this. Regarding if we need to change anything else on the infrastructure side, we are already using a preview version of the SDK to build and an even newer version of the compilers, so it should be straight forward to update to whichever version of the compilers we want that would include a specific feature.

shaggygi, 2019-06-02:
If I understood correctly, @MadsTorgersen mentioned at the C# Build 2019 session the default interface feature requires runtime changes.  This might not mean it requires .NET Core v3, but that's how I took it.  Anywhos, at least we can begin updating with other features for now.  We can close this now if you're good?

joperezr, 2019-06-03:
Yes, there are a bunch of features that do require the new runtime, but as you saw in  your switch PR there are some that won't require it. Sounds good, I'll close this for now, but feel free to reopen if you find any reason why we might want to retarget, since if such a case would exist, then we might think in cross-compiling for both netcoreapp2.1 and netcoreapp3.0

closed 2019-06-03 by joperezr