dotnet/iot #458: Implement better way to check for pin changes

chrishamm, 2019-05-29:
When a pin change event is registered, the GpioController starts polling pins in discrete time intervals which makes it difficult to use it for more or less time-critical applications. In my case, it is not called every time an event occurs. For my current application I need to react to such an event more than 1000 times per second and the GpioController seems to max out at approx. 230 pulses/sec, although direct polling via Read() without any delays tells me the pin level changes far more than 400 times per second. This problem is not present in WiringPi / Raspberry#.

I must admit I did not look into the C library sources in detail but perhaps a generic approach like this would be better in the long-term: https://github.com/torvalds/linux/blob/master/tools/gpio/gpio-event-mon.c AFAIK WiringPi supports pseudo-interrupts by interfacing the BCM's GPIO driver directly but perhaps the solution from that C program is already sufficient. AFAICS it only does "read" calls for event detection which are far cheaper than repeated polling.

joperezr, 2019-05-30:
Thanks for logging this @chrishamm. Would you mind trying your scenario using our Libgpiod driver? I believe that one should be more performant in terms of event handling so that might suit your needs. 
In order to try it, you would have to install libgpiod library by running [this script](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/libgpiod.sh) once in your board and then modifying your code just a bit so that when you are initializing your controller you use:
```c#
using (GpioController myController = new GpioController(PinNumberingScheme.Logical, new LibGpiodDriver()))
{
  //...
}
```

I still think it would be valuable to try to optimize our other drivers implementation (sysfs driver and RaspberryPi driver) for eventing, but this might get you unblocked for now.

chrishamm, 2019-06-13:
I'm sorry but I cannot test it because I have to deploy my software as a binary package and I don't want to ship kernel-dependent code. The code example I gave you uses only ioctl + file read for event detection and both are already being used in System.Devices.Gpio AFAIR. That's why I would really prefer an implementation based on that.

joperezr, 2019-06-13:
That's reasonable, I do know that our libgpiod-driver is much more efficient when it comes to event detection, but we should make sure that our other drivers are better. The code you pointed at is GPL license so we can't take a look at it and port it, but I'll leave this issue tracking the work of investigating how to implement the event detection mechanism with ioctl for the RaspberryPiDriver and the SysFSDriver.

Ellerbach, 2021-07-08:
[Triage] The way events are raised is different than what is described.

Also, improvements has been done thru PR #1567.

so closing this PR. If there is anything new, please feel free to reopen.

closed 2021-07-08 by Ellerbach