dotnet/iot #478: Error when trying to set speed of L298N H-Bridge (question how to)

MaklaCof, 2019-06-06:
I am trying to do a "HelloWord"  asp.net core application. I would like to control motor speed through web page.

But I don't know how to properly set motor speed. My code is:
```
        public async Task SetSpeed(int motor, double speed)
        {
            var sett = new DCMotorSettings()
            {
                Pin0 = 23,
                Pin1 = 24,
                PwmController = new PwmController(new SoftPwm()),
                PwmChip = 25,
                //PwmChannel = 1, // use for hardware PWM
                PwmFrequency = 50, // optional, defaults to 50
            };
            using (var m = DCMotor.Create(sett))
            { 
                m.Speed = speed / 100;
                Console.WriteLine($"Set speed of motor 1 to {m.Speed}.");
                await Task.Delay(3000);
            }
        }
```
With pin `23`, `24` and `25` I get error:

> Set speed of motor 1 to 0.6.
> dotnet :
>     + CategoryInfo          : NotSpecified: (:String) [], RemoteException
>     + FullyQualifiedErrorId : NativeCommandError
> 
> Unhandled Exception:
> 
> System.InvalidOperationException: Can not write to a pin that is not open.
>    at System.Device.Gpio.GpioController.Write(Int32 pinNumber, PinValue value)
>    at System.Device.Pwm.Drivers.SoftPwm.RunSoftPWM()
>    at System.Threading.ThreadHelper.ThreadStart_Context(Object state)
>    at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback,
> Object state)
> --- End of stack trace from previous location where exception was thrown ---
>    at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback,
> Object state)
>    at System.Threading.ThreadHelper.ThreadStart()
> 

I also try to set pin `33`, `35`, `37`, but I get

> System.Exception: Element not found.
> 
> Pin ' is not available. It is reserved by the system or in use for another function.
>    at Windows.Devices.Gpio.GpioController.OpenPin(Int32 pinNumber, GpioSharingMode sharingMode)
>    at System.Device.Gpio.Drivers.Windows10Driver.OpenPin(Int32 pinNumber)
>    at System.Device.Gpio.GpioController.OpenPin(Int32 pinNumber)
>    at System.Device.Pwm.Drivers.SoftPwm.OpenChannel(Int32 pinNumber, Int32 pwmChannel)
>    at System.Device.Pwm.PwmController.OpenChannel(Int32 pwmChip, Int32 pwmChannel)
>    at Iot.Device.DCMotor.DCMotor3Pin..ctor(PwmController pwmController, Double pwmFrequency, Int32 pwmChip, Int32 pwmChannel, Int32 pin0, Int32 pin1, GpioController controller)

I get pin numbers from [here](https://pi4j.com/1.2/pins/model-3b-plus-rev1.html).
What did I do wrong. 
I connected pins to my raspberry pi 3B+ like this:
[image]

[image]

joperezr, 2019-06-07:
@krwq do you mind taking a look here? From what I can tell, @MaklaCof is just following your sample from here:

https://github.com/dotnet/iot/blob/8f0a4a533867417844e72d045d66c1eb0e734ebc/src/devices/DCMotor/samples/Program.cs#L40-L51

so I would'nt expect the errors that he is seeing. 

> I get pin numbers from here.

I believe you are using Physical/Board pin numbering, and I believe that this binding is using Logical numbering. In order to get the right logical pin numbers, you can go to https://pinout.xyz/ and check the number that is becide BCM in order to get the right number. for example, for physical pin 23, it looks like the actual pin number is 11:

[image]

krwq, 2019-06-07:
@MaklaCof the issue looks like what @joperezr mentioned, if you still find any issues please make a picture or create Fritzing diagram of how you connected your motor, if there are still any issues after validating I will take a look if we somehow didn't regress this scenario.

MaklaCof, 2019-06-07:
Ok, fine thank you. 
I am not sure how wrongly connected pin triggers exception, but sound logic to me that I first try to connect correctly all the pins. I will update the thread in few hours when I get from work.

krwq, 2019-06-07:
@joperezr as part of this issue it might be worth checking what's our behavior when you try to open/write to the bogus pin

MaklaCof, 2019-06-09:
Ok here are findings:

- First I correctly connect pins, and code/motor start working.
- Then I set error handling [link](https://code-maze.com/global-error-handling-aspnetcore/)
Strange is that with my old code I get random error when API function end. (I think when dispose is called).
Sometimes I could call my API 10 times, sometimes 2 times ... Even stranger is, that application completely crash. I had to restart it. But not if throw exception in API.
- After that I change my code to save Motor in static variable:
```
    public class MotorsController : Controller
    {
        private static Dictionary<int, DCMotor> _motors = new Dictionary<int, DCMotor>();

        [HttpPut("SetSpeed/{motor}/{speed}")]
        public void SetSpeed(int motor, double speed, bool revert)
        {
            if (speed == 0)
            { 
                this.RemoveMotor(motor);
                return;
            }
            var m = this.GetMotor(motor);
            m.Speed = speed / 100 * (revert ? -1 : 1);
            Console.WriteLine($"Set speed of motor 1 to {m.Speed}.");
        }

        private DCMotor GetMotor(int motorIndex)
        {
            var sett = new DCMotorSettings()
            {
                Pin0 = 13,
                Pin1 = 19,
                PwmController = new PwmController(new SoftPwm()),
                PwmChip = 26,
                PwmFrequency = 50
            };

            if (!_motors.TryGetValue(motorIndex, out var motor))
            {
                motor = DCMotor.Create(sett);
                _motors.Add(motorIndex, motor);
            }
            return motor;
        }

        private void RemoveMotor(int motorIndex)
        {
            if (_motors.TryGetValue(motorIndex, out var motor))
            {
                motor.Dispose();
                _motors.Remove(motorIndex);
            }
        }
    }
```
Every time I call dispose, I get exception and my application completely crash. Witch again is really strange. Maybe something to look for @joperezr  (Because throwing an error in application doesn't crash application.)
- Then I try to set speed to 0, before calling dispose. Exception happen random (after 2 tries, 5 tries ...)
- Then I set speed to 0, wait 1 second, call dispose and now it seems it works.
```
        private async Task RemoveMotor(int motorIndex)
        {
            if (_motors.TryGetValue(motorIndex, out var motor))
            {
                motor.Speed = 0;
                await Task.Delay(1000);
                motor.Dispose();
                _motors.Remove(motorIndex);
            }
        }
```
Not sure @krwq if calling dispose should set motor speed to 0, and wait a few moments to avoid exceptions. I will leave that to you.

Now I just have one question @krwq  that probably it is not related to the code. Even if I set speed to 0.01, motor ([Lego](https://shop.lego.com/en-US/product/LEGO-Power-Functions-M-Motor-8883)) runs approximately at half speed. If I further decrease speed, motor rotation is not slower but it stop for short moment and run again and stop for a short moment and run again ....   
Is this problem with PwmFrequency or L298N H-Bridge itself? I get a lot better rotation (continuous rotation) if I drop voltage. Is this normal? I don't known anything about PWM, but it seems that this works a lot worse comparing to voltage change.

krwq, 2019-06-10:
> Every time I call dispose, I get exception and my application completely crash

Do you call dispose at any other time than end? is it possible that some code still runs after dispose (i.e. thread running, setting speed or something)

How does it crash? Are you getting any exception message? (can you copy the whole output related to exception?)

Especially randomness sounds like something running in a different thread - I'd expect this to be rather deterministic unless the code related to GPIO has an issue (hence the message would be useful).

> Even if I set speed to 0.01, motor (Lego) runs approximately at half speed

When I think about it now I think the name I've used might be a bit misleading.
Speed is actually relative voltage (or more accurately PWM fill but considering motor has inertia they are close to each other) fed onto the engine (0 is none, 1 is maximum).
Considering static friction and other physical properties this is not exactly speed but on the other hand I do not have idea for better term to use.

In order to get something which is translatable to speed you'd need to feed inputs to the motor class into the PID controller and find the right properties (currently we do not have code for that in here).

MaklaCof, 2019-06-10:
About number 1.
This is true, but not happening right now, since I am the only one triggering operations on dictionary.

About number 2.
This is not true. Controller is disposed immediately when API function is executed. In my case, that means that exception is triggered and motor stops.

Error is:

> 
> Disposing motor 1.
> C:\Program Files\dotnet\dotnet :
>     + CategoryInfo          : NotSpecified: (:String) [], RemoteException
>     + FullyQualifiedErrorId : NativeCommandError
> 
> Unhandled Exception:
> 
> System.InvalidOperationException: Can not write to a pin that is not open.
>    at System.Device.Gpio.GpioController.Write(Int32 pinNumber, PinValue value)
>    at System.Device.Pwm.Drivers.SoftPwm.RunSoftPWM()
>    at System.Threading.ThreadHelper.ThreadStart_Context(Object state)
>    at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback,
> Object state)
> --- End of stack trace from previous location where exception was thrown ---
>    at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback,
> Object state)
>    at System.Threading.ThreadHelper.ThreadStart()

This lego motor are more or less standard ([image](http://www.philohome.com/pf/diagram.gif) from [here](http://www.philohome.com/pf/pf.htm)).  
[Wire hacking](https://scuttlebots.com/2014/03/02/lego-pf-hacking-wiring/)

I also using the same connector when plugged to 6V power adapter. 

We can forget about speed, I agree there is no better naming. I didn't mean that speed would actually mean speed, but I do though, that setting speed to 0.5 would mean approximately half of full speed.

For now I am happy with hot it works. I know I need to set motor speed to 0 and wait a few moments before `Motor` object is disposed. And it works. I was hoping to get smoother rotating with low speed, but it will be ok.

Thanks.

krwq, 2019-07-30:
@MaklaCof I think I encountered this issue when testing fix for #641. I included the patch in the PR.

The issue only happens in 1 or 2 pin mode when using software PWM which uses the same controller for PWM and motor direction

PR #643 (2019-08-07): Simplify DCMotor abstraction
Fixes: #641
Fixes: #478

See details in #641

Also fixed the dispose issue where DCMotor in 2/1 pin mode is using software PWM with the same controller and controller gets disposed before software PWM stops writting (which surfaces as exception as described in #478) - noticed when testing.

cc: @shaggygi @MaklaCof
closed 2019-08-07 by krwq