dotnet/iot #491: .Net Core on Raspberry Pi 3 A+

JawadJaber, 2019-06-11:
It is [stated](https://github.com/dotnet/iot/tree/master/samples) that .Net Core does not work on RP models A. But as per [this](https://www.modmypi.com/blog/raspberry-pi-comparison-table), RP model 3A+ is with ARMv8-A which is same as RP 3B model. So please confirm or revise the page if necessary.

joperezr, 2019-06-11:
@JawadJaber thanks for catching that, do you have a Pi 3A+? We don't currently have one to test with, so if you have validated that .NET Core runs well on it then we can go and fix the docs so that we mention that A+ should also work.

shaggygi, 2019-06-12:
@joperezr I purchased one of these a few weeks back, but never got around setting up.  I installed Raspbian tonight and was able to write values using DeviceApiTester with GPIO commands.  I haven't tried other types like I2C and SPI yet, but I'd bet those work, as well.  I'll try to do some basic write tests with those this weekend to double-check.

JawadJaber, 2019-06-12:
@joperezr I ordered Pi 3A+ and is not delivered yet, I have dotnet core that is under production with Pi 3B, 3B+. Once received, I will deploy my dotnet core for testing and inform you.

joperezr, 2019-06-12:
@shaggygi @JawadJaber thanks so much to both of you for checking. Once you confirm I'll go ahead and update our docs. The fact that .NET Core runs fine for @shaggygi makes me think that there shouldn't be any problems but it would be great to just double check I2C and SPI if possible 😄

krwq, 2019-06-18:
Raspberry Pi 3 A+ work correctly on my PIs.

joperezr, 2019-06-18:
Great, I'll leave this open though in order to track the change in the docs to say we do work there properly.

PR #515 (2019-06-21): Included RPi 3 A+ as supported
Fixes #491 

Updated wording the RPi 3 Model A+ is supported.
commit 33432162 (2019-06-21): Included RPi 3 A+ as supported (#515)

* Included RPi 3 A+ as supported

Fixes #491 

Updated wording the RPi 3 Model A+ is supported.

* Update README.md

* Update README.md
closed 2019-06-21 by krwq