dotnet/iot #498: SenseHat does not create appropriate I2C devices

krwq, 2019-06-13:
Problem is that we do not have a generic way to create I2C/SPI on Windows and Linux (i.e. I2cDevice.Create(..)).

I.e. https://github.com/krwq/iot/blob/master/src/devices/SenseHat/SenseHatLedMatrixI2c.cs#L142

this is causing that users have to use workarounds in order to use SenseHat.

cc: @joperezr

shaggygi, 2019-06-13:
Referencing https://github.com/dotnet/iot/issues/118#issuecomment-496053073

krwq, 2019-06-13:
I'm seeing 2 options:
- long term: System.Device: SpiDevice.Create/I2cDevice.Create
- short term: Iot.Device.Binding: GenericSpiDevice/GenericI2cDevice (put it somewhere in the common folder)

joperezr, 2019-06-13:
I agree with @shaggygi, this is a dupe of #118 and honestly I wouldn't rush adding any of the two solutions, and instead implement a proper SpiController and I2cController. I think adding these is not blocking 3.0 as what our bindings currently do is to take a SpiDevice or a I2cDevice in their constructor, but it would be a nice to have. I'm closing this as a dupe, but @krwq feel free to add your suggestions into #118

shaggygi, 2019-06-17:
This should actually be reopened even though this discussion is related to ideas how to fix the issue.

We can fix/update the SenseHat binding and close this issue after #118 is completed.

closed 2019-06-27 by krwq