dotnet/iot #508: TCA9548 Low-Voltage 8-Channel I 2C Switch

shaggygi, 2019-06-18:
@joperezr Go ahead an assign this to me, as well.  I have a couple on order for a rainy day :wink: :wink: 

Soon after the `I2cController` is complete and added to core API, it would be a good exercise to create a binding that implements `II2cController`.

The TCA9548 is basically an I2C multiplexer that branches out to more I2C buses.
**Datasheet**: http://www.ti.com/lit/ds/symlink/tca9548a.pdf

Disclaimer: Here is some **very** early thinking, but not completely sure if this is how it would work in the end.  Adding as scratchpad work for now.

```csharp
// Tca9548  Class
public class Tca9548 : II2cController

// Tca9548  Constructor
public Tca9548(I2cConnectionSettings settings, II2cController? controller = null)

// Create master controller (e.g. RPi3 with Linux, etc.).
var rpiController = new I2cController()
{
    ShouldDispose = false // The default is true, so we instruct to not dispose it downstream.
};

// Create Tca9548 with master controller
var tca9548 = new Tca9548(new I2cConnectionSettings(1, 0x33), rpiController)
{
    ShouldDispose = true  // This is the default value, but showing for demo.  Will dispose downstream.
}

// Create and add  Mcp23008 to Tca9548 controller.
// Note this is currently not how the Mcp23008 is today as it uses I2cDevice.
// Since an I2cDevice wasn't created for the mcp23008 and opened, this creates default I2cDevice with ShouldDispose = true.
var mcp23008 = new Mcp23008(new I2cConnectionSettings(5, 0x20), tca9548);

// Party on the Mcp23008 🎉 
mcp23008.OpenPin(3, PinMode.Output);
mcp23008.Write(3, PinValue.High);  // RPi3 writes to Tca9548 which writes to Mcp23008.

// This disposes mcp23008 I2cDevice, disposes tca9548 and removes it from the rpiController.
mcp23008.Dispose();

// Since it was set to not dispose, you can open/close more devices or go ahead and dispose if finished using.
rpiController.Dispose();
```

shaggygi, 2019-06-23:
There's more devices in the family, so structure appropriately. 

https://www.nxp.com/docs/en/application-note/AN262.pdf

krwq, 2023-02-09:
[Triage] It's already added https://github.com/dotnet/iot/tree/main/src/devices/Tca954x

closed 2023-02-09 by krwq