dotnet/iot #519: UnixI2cDevice.Linux.cs Calls Initialize each time

shaggygi, 2019-06-22:
In the UnixI2cDevice.Linux.cs `I2cDevice` class, it calls Initialize method every time for Reads/Writes.  Should this be called once when instantiating object or is there more low-level things going on that it needs to check?

krwq, 2019-06-28:
we should call it once I think. If user creates a device they will most likely use it so not much benefit from delaying that.

I'm not sure I understand second paragraph of this issue.

shaggygi, 2019-06-29:
The class has a default path...

`private const string DefaultDevicePath = "/dev/i2c";`

I'm assuming this is where I2C low-level code is installed on Linux.  Each time it calls Initialize method, it gets the path from DevicePath which the default path and bus ID.  I guess one could instantiate the I2cDevice object and later change the DevicePath if needed.

I just  don't see the need to change the DevicePath.  Also, you wouldn't be able to change when auto-creating based on new Create scheme.  Just was thinking we could move method to initialize once when creating the object and not take a hit on each read/write.

krwq, 2019-06-29:
@shaggygi yes, we don't need to change that (also there is chroot)

Ellerbach, 2021-07-08:
[Triage] Closing this issue as it is now addressed with the introduction of the [I2C Bus](https://github.com/dotnet/iot/blob/main/src/System.Device.Gpio/System/Device/I2c/UnixI2cBus.cs). The device is open only once when the bus is created. A new device will just be added to the existing bus.

closed 2021-07-08 by Ellerbach