dotnet/iot #539: using /sys/class/leds

WattsC-90, 2019-07-01:
I have some linux hardware, looking at the code and through tests, it defaults to the sysfs driver. 
The hardware itself has some normal IO available in /sys/class/gpio, but these are the purest forms of IO. Is there anyway using the library currently, to control the LED devices pointed to in the sys/class/leds folder? 

I am just trying some code out on a yocto linux board but cant see how it is possible at present (without writing extra code)

krwq, 2019-07-01:
@WattsC-90 do you mean the green and red LEDs? I don't believe we currently have a library for that. Would you be interested in adding one?

krwq, 2019-07-01:
other option would be to explore whole /sys/firmware/devicetree/base and create smaller workitems like this one from this one

WattsC-90, 2019-07-01:
yes i was just looking at that.. perhaps a device/IO controller with different supported classes/types (io/led/analogue(?)) or maybe just extend GPIO to support Analogue/LED types.

I would love to say yes to helping to add one, but struggling to find the time to sleep at the moment let alone take on more stuff! I will have a look when im playing around here and there... maybe something easy will jump out!

krwq, 2019-07-01:
@WattsC-90 let's figure out how we want this exposed first, I think the implementation will be rather simple.

I'm not sure this should be directly on GpioController (unless we figure out the whole device tree). Instead I think we should do something similar to what [CpuTemperature](https://github.com/dotnet/iot/tree/master/src/devices/CpuTemperature) did

joperezr, 2019-07-01:
Is that path only there to control the hardware (board's) LEDs? If so, I suppose this path isn't there in all different boards? If this is very specific only to the Pi I'm not sure if it makes much sense to add. Is there an advantage of using /sys/class/leds over /sys/class/gpio to control those LEDs?

WattsC-90, 2019-07-01:
@joperezr  Yes it's a hardware implementation. I am not using a PI, I am using a different Linux board which has two leds mapped into the /sys/class/leds folder.

> Is there an advantage of using /sys/class/leds over /sys/class/gpio to control those LEDs?

I agree that there is no advantage to leds over gpio, but this is a hardware platform that I have been given and have no control on, unfortunately dont have it on a breadboard, only a PCB in a box with a lot of connectors. First stages were to read a pin and flash an LED! 

@krwq  I think you're probably right, I am trying to think of a generic solution that could be extended to support multiple items in the device tree but probably over the top/over-engineered..

krwq, 2019-07-01:
@joperezr presumably different boards will have different LEDs.

I.e. on my PC I can see more LEDs:

```
/sys/class/leds$ ls
input28::capslock  input28::numlock  input28::scrolllock  input34::capslock  input34::numlock  input34::scrolllock
```

we should presumably detect all of them and can read names from i.e.:

```
/sys/class/leds/input28::capslock/device$ cat name 
Compx 2.4G Receiver
```

krwq, 2019-07-01:
@WattsC-90 I think solution for this should be LED specific. So i.e.:

```csharp
class CpuLed
{
  public string Name { get; }
  public int Brightness { get; set; }
  public int MaxBrightness { get; }
  public static IEnumerable<CpuLed> EnumerateLeds();
}
```

joperezr, 2019-07-01:
> I think solution for this should be LED specific.

If there is no real advantage from using /sys/class/gpio, I would not add this and instead just us gpio to control them. It is easier to think of them as just another LED than to try to over-engineer new types for this IMO.

WattsC-90, 2019-07-01:
@joperezr The LEDs cant be controlled from the current GpioController, are you suggesting extending it to handle different /sys/class/*x* types?

joperezr, 2019-07-01:
No, what I was suggesting is that if this LEDs can be controlled with /sys/class/gpio/ we should just use that

WattsC-90, 2019-07-01:
it cant, thats the issue. It exists in a different area of /sys/class, where the GPIO is in /sys/class/gpio, the leds are in /sys/class/leds

[image]

versus the /sys/class/gpio:
[image]

joperezr, 2019-07-01:
They may have a different name, there are some cases in some boards where the hardware LEDs are mapped to a gpio, for example, for the Raspberry Pi, the Green OK onbard LED is mapped to Gpio 16:

```
The green OK LED is hardwired to GPIO 16 and is programmed in the 'firmware' to indicate SD card activity...
```

WattsC-90, 2019-07-01:
oh i see, I am not sure how I would find that out.. good idea though! I will go see what i can scrape up

krwq, 2019-07-01:
@joperezr assuming that is possible I'm not sure if it's a good idea to bypass the driver - I suspect it may cause some really painful issues

krwq, 2019-07-03:
I think we should start with exposing this as the device and then later we can move to System.Devices.Gpio.

@WattsC-90 I feel like this is a good addition I'd recommend sending PR to add a version of this which looks reasonably good to src/devices

WattsC-90, 2019-07-03:
@krwq OK, I am busy for the next couple of weeks but I will have a crack at it after then..

PR #738 (2019-09-29): Add on-board LED controller
fix #539
closed 2019-09-29 by krwq