dotnet/iot #553: Local builds are not clean

krwq, 2019-07-03:
I've got similar errors on both Windows and Linux and both of my builds were clean (`git clean -fdx` + clean nuget cache)

```
  Ws28xx -> /home/krwq/work/iot/src/devices/Ws28xx/bin/Debug/netcoreapp2.1/Ws28xx.dll
  Ws28xx.Samples -> /home/krwq/work/iot/src/devices/Ws28xx/samples/bin/Debug/netcoreapp2.1/Ws28xx.Samples.dll
/home/krwq/.nuget/packages/microsoft.sourcelink.common/1.0.0-beta2-19324-01/build/Microsoft.SourceLink.Common.targets(50,5): error : Source control information is not available - the generated source link is empty. [/home/krwq/work/iot/src/System.Device.Gpio/System.Device.Gpio.csproj]
/home/krwq/.nuget/packages/microsoft.sourcelink.common/1.0.0-beta2-19324-01/build/Microsoft.SourceLink.Common.targets(50,5): error : Source control information is not available - the generated source link is empty. [/home/krwq/work/iot/src/System.Device.Gpio/System.Device.Gpio.csproj]
  System.Device.Gpio -> /home/krwq/work/iot/artifacts/bin/System.Device.Gpio/Debug/netstandard2.0/linux/System.Device.Gpio.dll
  System.Device.Gpio -> /home/krwq/work/iot/artifacts/bin/System.Device.Gpio/Debug/netstandard2.0/win/System.Device.Gpio.dll
/home/krwq/.nuget/packages/microsoft.sourcelink.common/1.0.0-beta2-19324-01/build/Microsoft.SourceLink.Common.targets(50,5): error : Source control information is not available - the generated source link is empty. [/home/krwq/work/iot/src/System.Device.Gpio.Tests/System.Device.Gpio.Tests.csproj]
  System.Device.Gpio.Tests -> /home/krwq/work/iot/artifacts/bin/System.Device.Gpio.Tests/Debug/netcoreapp2.1/win-arm/System.Device.Gpio.Tests.dll
/home/krwq/.nuget/packages/microsoft.sourcelink.common/1.0.0-beta2-19324-01/build/Microsoft.SourceLink.Common.targets(50,5): error : Source control information is not available - the generated source link is empty. [/home/krwq/work/iot/src/System.Device.Gpio.Tests/System.Device.Gpio.Tests.csproj]
  System.Device.Gpio.Tests -> /home/krwq/work/iot/artifacts/bin/System.Device.Gpio.Tests/Debug/netcoreapp2.1/linux-arm/System.Device.Gpio.Tests.dll
/home/krwq/.nuget/packages/microsoft.sourcelink.common/1.0.0-beta2-19324-01/build/Microsoft.SourceLink.Common.targets(50,5): error : Source control information is not available - the generated source link is empty. [/home/krwq/work/iot/src/System.Device.Gpio/System.Device.Gpio.csproj]
  System.Device.Gpio -> /home/krwq/work/iot/artifacts/bin/System.Device.Gpio/Debug/netstandard2.0/notsupported/System.Device.Gpio.dll
  device-listing -> /home/krwq/work/iot/tools/device-listing/bin/Debug/netcoreapp2.1/device-listing.dll
  DeviceApiTester -> /home/krwq/work/iot/tools/DevicesApiTester/bin/Debug/netcoreapp2.1/DeviceApiTester.dll
  _DeviceBinding -> /home/krwq/work/iot/tools/templates/DeviceBindingTemplate/dotnet_new_device-binding_csharp/bin/Debug/netcoreapp2.1/_DeviceBinding.dll
  _DeviceBinding.Samples -> /home/krwq/work/iot/tools/templates/DeviceBindingTemplate/dotnet_new_device-binding_csharp/samples/bin/Debug/netcoreapp2.1/_DeviceBinding.Samples.dll
  _DeviceBinding_Not_Required.Tests -> /home/krwq/work/iot/tools/templates/DeviceBindingTemplate/dotnet_new_device-binding_csharp/tests/bin/Debug/netcoreapp2.1/_DeviceBinding_Not_Required.Tests.dll
  force-sensitive-resistor -> /home/krwq/work/iot/samples/force-sensitive-resistor/bin/Debug/netcoreapp2.1/force-sensitive-resistor.dll
  led-blink -> /home/krwq/work/iot/samples/led-blink/bin/Debug/netcoreapp2.1/led-blink.dll
  led-more-blinking-lights -> /home/krwq/work/iot/samples/led-more-blinking-lights/bin/Debug/netcoreapp2.1/led-more-blinking-lights.dll

Build FAILED.

/home/krwq/.nuget/packages/microsoft.sourcelink.common/1.0.0-beta2-19324-01/build/Microsoft.SourceLink.Common.targets(50,5): error : Source control information is not available - the generated source link is empty. [/home/krwq/work/iot/src/System.Device.Gpio/System.Device.Gpio.csproj]
/home/krwq/.nuget/packages/microsoft.sourcelink.common/1.0.0-beta2-19324-01/build/Microsoft.SourceLink.Common.targets(50,5): error : Source control information is not available - the generated source link is empty. [/home/krwq/work/iot/src/System.Device.Gpio/System.Device.Gpio.csproj]
/home/krwq/.nuget/packages/microsoft.sourcelink.common/1.0.0-beta2-19324-01/build/Microsoft.SourceLink.Common.targets(50,5): error : Source control information is not available - the generated source link is empty. [/home/krwq/work/iot/src/System.Device.Gpio.Tests/System.Device.Gpio.Tests.csproj]
/home/krwq/.nuget/packages/microsoft.sourcelink.common/1.0.0-beta2-19324-01/build/Microsoft.SourceLink.Common.targets(50,5): error : Source control information is not available - the generated source link is empty. [/home/krwq/work/iot/src/System.Device.Gpio.Tests/System.Device.Gpio.Tests.csproj]
/home/krwq/.nuget/packages/microsoft.sourcelink.common/1.0.0-beta2-19324-01/build/Microsoft.SourceLink.Common.targets(50,5): error : Source control information is not available - the generated source link is empty. [/home/krwq/work/iot/src/System.Device.Gpio/System.Device.Gpio.csproj]
    0 Warning(s)
    5 Error(s)
```

krwq, 2019-07-03:
FYI `export EnableSourceLink=false` workarounds the issue

joperezr, 2019-07-03:
I haven't been able to repro this. If you can get a binlog I can help investigate.

krwq, 2019-07-05:
I've sent you binlog offline

joperezr, 2019-07-05:
From what I can see, looks like the remotes on your repo are not set correctly. Source link will try to find the remote source in order to construct the url for source linking. I would recommend cloning again, and then do a clean build, It shouldn't repro in that case.

krwq, 2019-07-05:
@joperezr what do you mean remotes are not set correctly? Is there any specific name of the remote I should have to make that work? Do I need to have exactly 1 remote? This sounds like an arcade bug, it should not force me to any specific git setup.

joperezr, 2019-07-05:
nope, you can have multiple. I believe as long as you have one that is called origin then you should be fine, although I might be wrong.

krwq, 2019-07-05:
@joperezr I call it `dotnet` this sounds like an arcade bug to me.

joperezr, 2019-07-05:
I'm not familiar enough with source link, so I would recommend the following in order to see if this is the source of the issue and if so log a bug with arcade:

1. Create a new folder in your disk and clone iot repo.
2. Without changing the name of the remote, call build.sh. (build should succeed)
3. Rename the remote, clean your enlistment and try to build again. 

If step 3 failed, then looks like that is the issue, so it would be fine to log a suggestion in arcade.

krwq, 2019-07-05:
@joperezr confirming, my build is failing when my remote is named `dotnet` and is successful when it is called `origin`. cc: @dotnet/dnceng

joperezr, 2019-07-05:
How did you rename it? have you tried using `git remote rename origin dotnet`? that might update all that is needed in order for sourcelink to work.

dagood, 2019-07-05:
Filed https://github.com/dotnet/arcade/issues/3236 for this, I hit it in Core-Setup as well.

krwq, 2019-09-04:
@tmat I've seen you closed https://github.com/dotnet/arcade/issues/3236 - was the issue with remotes fixed or do you still need a specific remote name for build to work?

krwq, 2019-09-04:
Closing assuming yet - please re-open if this is still required

closed 2019-09-04 by krwq