dotnet/iot #556: Build issues

RheaAyase, 2019-07-03:
Thought that I'll give [PR#540](https://github.com/dotnet/iot/pull/540) a shot, encountered these issues while trying to build it.

1) global.json is referencing an invalid SDK version (2.2.203 is Visual Studio toolset version and is not a valid .NET Core SDK version - that would be 2.2.103)

2) build.sh script attempts to install some obscure version of SDK on my system (and therefor probably wreck my own distro packaged sdk) even when it has correctly set `useInstalledDotNetCli`

3) After disposing of the above issues I'm getting: `Mcp25xxx.Sample.cs(7,25): error CS0234: The type or namespace name 'Drivers' does not exist in the namespace 'System.Device.Spi' (are you missing an assembly reference?) [src/devices/Mcp25xxx/samples/Mcp25xxx.Samples.csproj]`

(Neither is a blocker, but they're all inconveniences. And I wouldn't dare tell you how to resolve these as I am not the active user for it.)

joperezr, 2019-07-03:
Hey @RheaAyase thanks for logging this.

> global.json is referencing an invalid SDK version (2.2.203 is Visual Studio toolset version and is not a valid .NET Core SDK version - that would be 2.2.103)

you don't need to have this installed manually, this will be installed as part of the build inside the /iot/.dotnet/ folder

> build.sh script attempts to install some obscure version of SDK on my system (and therefor probably wreck my own distro packaged sdk) even when it has correctly set useInstalledDotNetCli

It won't install it globally in your machine, it will only install it in the `.dotnet` folder in the root of the repo and then it will use that one to build the product. If you clean the repo (via `git clean -fdx` from the root) or delete the folder then it won't this SDK won't be in your machine.

> After disposing of the above issues I'm getting: Mcp25xxx.Sample.cs(7,25): error CS0234: The type or namespace name 'Drivers' does not exist in the namespace 'System.Device.Spi' (are you missing an assembly reference?) [src/devices/Mcp25xxx/samples/Mcp25xxx.Samples.csproj]

This problem has been fixed recently with https://github.com/dotnet/iot/commit/d6274e441c0040a70275b5acff4c3bbb625053ce#diff-d0db498d49445da1089f9b26f1fba292R9 so you won't see this any longer.

Given I took this issue as a 3-part question which I believe I answered I'm closing it, but please do feel free to reopen if you have any follow up questions/comments.

closed 2019-07-03 by joperezr

krwq, 2019-07-03:
@RheaAyase yes, this has been a pain for quite a bit and we never had a chance to fix it. Here is the workaround:

```bash
# build repo first to make sure iot gets its own version of .NET
./build.sh

# This will cause that using dotnet exe will not look at any other sdks rather than the one next to that dotnet exe.
export DOTNET_MULTILEVEL_LOOKUP=0
# Makes sure REPO_ROOT/.dotnet/dotnet is on your path, adjust per your paths
export PATH=$HOME/src/iot/.dotnet:$PATH
```

after that using `dotnet` should work

RheaAyase, 2019-07-03:
I don't think that this is answered, especially point 1)

> > global.json is referencing an invalid SDK version (2.2.203 is Visual Studio toolset version and is not a valid .NET Core SDK version - that would be 2.2.103)
> 
> you don't need to have this installed manually, this will be installed as part of the build inside the /iot/.dotnet/ folder

Not needing it installed is different from it being wrong SDK version to be referenced. 2.2.10x is the actual SDK version, while 2.2.20x is just extra tooling for Visual Studio 19.0 (and similarly 2.2.30x is for VS 19.1) and as such, these are not there for Linux systems (these are not source-built)

I don't know a reason for this to be referenced, I don't think that there is one, but feel free to enlighten me.

I know I'm nitpicking here but eh... What I don't understand though is the whole `build.sh` and it's existence. One liner would suffice, or even nothing at all. We don't need all the logic that installs SDK, etc etc... And if all that is there for some CI stuff, then it should be labeled as such, and should not be enabled by default because you set CI up once and forget it. Being accessible for new open source contributors is more important =)

joperezr, 2019-07-03:
I'm not sure where you are seeing the invalid SDK version in our global.json, we are currently referencing 3.0 preview 6 sdk:

https://github.com/dotnet/iot/blob/0009b06cfe718e78c05ff625a16f1e174dcb92f5/global.json#L2-L12

Have you tried pulling the latest from master and trying again? 

> What I don't understand though is the whole build.sh and it's existence. 

In order to build our libraries, we rely on a shared infrastructure (source of the infrastructure is https://github.com/dotnet/arcade) which all dotnet repos use(like corefx and coreclr). We use this infrastructure for much more than just CI, for example, we use it in order to define all of the Versioning, licensing, testing, and packaging infrastructure that we require to build. This centralized infrastructure also controls the version of the SDKs and compilers we use, so it would also update that so that all of our repos (all of the .NET Core product) build using the same compilers and use the same versioning scheme. build.sh is a script we use in order to bootstrap all of that shared infrastrcuture, restore the right sdks, and download all of the tools that we will need at build time, so that your local builds would look the same as our official ones. If there is a particular part of the script that you don't understand or you don't agree with, I'm happy to chat to try to see how we can make things better 😄

krwq, 2019-07-03:
@joperezr I'm on @RheaAyase, basically every new person touching iot repo hits this or similar issue

RheaAyase, 2019-07-04:
Interesting, even I, building every commit in corefx, never realised that this is part of the same infra, I just saw this as a random nuget library that isn't related. That being the case though, this issue is even bigger and spans across all the repositories that have been _assimilated into the collective._

I don't know, it just seems weird to have a regular dotnet project depend on all this stuff. Perhaps better clarifying that for a simple scratch build, even `dotnet build` is sufficient and `build.sh` is not necessary? Documenting this and perhaps other hints and tips on how to build it and contribute in the CONTRIBUTING.md could go a long way to help out potential contributors.

I think that this is more important in this repository than others because of it's hobby/geeky use-case.

joperezr, 2019-07-05:
We have an issue #100 that will track the addition of detailed instructions on how to build and how to contribute to the repo along with other documentation.

krwq, 2019-07-05:
@joperezr would it perhaps make sense to add some kind of env.sh/env.ps1 which would be script meant to be sourced, add correct dotnet to path, disable source link, first time dotnet experience and source linking? This way users can use dotnet smoothly with whatever setup they have and if anyone has any issues with build you just tell them to source `env.sh`. Perhaps this should be done in dotnet/arcade itself

joperezr, 2019-07-05:
Sourcelink shouldn't be disabled, I think that's an issue that only you saw in your machine and IMO it is due to a problem with your enlistment. As to generating an env.sh, that is doable although it would be very small (two lines, add the repo local dotnet to the path and disable first time experience) so I'm not sure if that would be worth it.