dotnet/iot #559: Add Bindings for Adafruit STEMMA Soil Sensor OR Adafruit Seesaw

andersonimes, 2019-07-04:
I'm trying to use a Adafruit STEMMA Soil Sensor - I2C Capacitive Moisture Sensor to create a garden soil monitor. I2C interfaces are a bit over my head and a helper binding would be great. Adafruit has this specific sensor package, but the combination is supported by the slightly more generic Seesaw library that they have created (I think that's what I'm reading). Porting either of those over would be pretty helpful.

andersonimes, 2019-07-04:
https://learn.adafruit.com/adafruit-stemma-soil-sensor-i2c-capacitive-moisture-sensor/overview

Frankenslag, 2019-07-06:
I would like to have a look at this.... I think that in reality if a seesaw binding is created then reading the soil sensor is a fairly minor exercise. I will need a little time to source the seesaw breakout so please give me a week or two.

Frankenslag, 2019-08-01:
@andersonimes 

Please see PR #648 This should allow you to use the soil sensor as below  assuming that you connect it to I2C bus 1 ...

    const byte Adafruit_Seesaw_SoilSensor_I2cAddress = 0x36;
    const byte Adafruit_Seesaw_SoilSensor_I2cBus = 0x1;

    using (I2cDevice i2cDevice = I2cDevice.Create(new I2cConnectionSettings(Adafruit_Seesaw_SoilSensor_I2cBus, Adafruit_Seesaw_SoilSensor_I2cAddress)))
    using(Seesaw ssDevice = new Seesaw(i2cDevice))
    {
        while(true)
        {
            Console.WriteLine($"Temperature: {ssDevice.GetTemperature()}'C");
            Console.WriteLine($"Capacitive: {ssDevice.TouchRead(0)}");
            ssDevice.SetGpioPinMode(1, PinMode.Output);
            System.Threading.Tasks.Task.Delay(1000).Wait();
        }
    }

krwq, 2023-02-09:
[Triage] This is partially addressed already, we're also not tracking binding requests anymore. Please send PR. If there is anything to discuss here please reopen

closed 2023-02-09 by krwq