dotnet/iot #565: Proper way to comment constructor

shaggygi, 2019-07-04:
I came across this topic and wasn't sure if we should be doing similar in this repo.

https://stackoverflow.com/questions/6868138/whats-the-proper-way-to-comment-a-constructor-in-a-generic-class

Currently...

```csharp
/// <summary>
/// Initializes new instance of SpiConnectionSettings.
/// </summary>
/// <param name="busId">The bus ID the device is connected to.</param>
/// <param name="chipSelectLine">The chip select line used on the bus.</param>
public SpiConnectionSettings(int busId, int chipSelectLine)
```

Should the summary be...

```csharp
/// <summary>
/// Initializes a new instance of the <see cref="SpiConnectionSettings"/> class.
/// </summary>
/// <param name="busId">The bus ID the device is connected to.</param>
/// <param name="chipSelectLine">The chip select line used on the bus.</param>
public SpiConnectionSettings(int busId, int chipSelectLine)
```

krwq, 2019-07-04:
@shaggygi if you mean `<see cref="SpiConnectionSettings"/>` then ideally yes but better not to have it than have no doc at all

shaggygi, 2019-07-04:
Okay.  Just checking what the norm is as if it was preferred I could do a PR to update throughout in this format.  Lemme know and I can allocate some time.

cc: @joperezr

joperezr, 2019-07-05:
Yes, as @krwq mentions idieally we would like to have `cref` like documentation. We will eventually want to upload all of our comments (for System.Device.Gpio) to MSDN docs, where they will be audited and so suggestions like using cref will be pointed out, so getting  ahead of them would be great 😄

shaggygi, 2019-07-05:
@joperezr sounds good.  i'll try to do a first round of this over the weekend.

PR #569 (2019-07-13): Updated summaries using cref with constructors
Fixes #565

This focused on updating files that already included similar wording.  Did not update non-OS files (Linux > Windows, vice versa), internal classes or PWM (since they are about to change).
closed 2019-07-13 by krwq