dotnet/iot #580: Add support for /dev/mem to the RaspberryPiDriver

joperezr, 2019-07-09:
When testing our support for arm64, I discovered that the Debian available for arm64 doesn't have a /dev/gpiomem memory mapped file which is what the RaspberryPiDriver uses today in order to get access to the registers. Instead, it only has the /dev/mem memory mapped file, which should provide access to such registers. We should add the capability to the RaspberryPi Driver that in case we can't find /dev/gpiomem/ we should be able to just use the /dev/mem file instead. That would make the RaspberryPiDriver work for arm64 on this specific distro.

Frankenslag, 2019-07-12:
@joperezr  I would like to have an informal look at this. Did you get the image from https://wiki.debian.org/RaspberryPi3 ?

joperezr, 2019-07-12:
@Frankenslag Awesome! yes, more specifically the image I got was from: `https://people.debian.org/~gwolf/raspberrypi/20190628/20190628_raspberry-pi-3_buster_PREVIEW.img.xz`. That said, you shouldn't need that to prototype this since /dev/mem is also available in regular Raspbian arm32 images. We initially picked /dev/gpiomem because of two reasons: 1) You only get access to the gpio related registers, which contained all we need at the time, and mor importantly 2) It didn't require to run as sudo, whereas /dev/mem does. This means that when using the RaspberryPi drivier today, your app doesn't necessarily need to run as sudo which would change with /dev/mem. The ideal implementation, would use /dev/gpiomem when available, and fallback to /dev/mem if not found. I did start adding this support some time ago but never finished, here is a pointer to my partial work in case it helps: https://github.com/joperezr/corefxlab/blob/970f4a57047e53ab22ec972e303307c0afca745d/src/System.Devices.Gpio/RaspberryPiDriver.cs

Frankenslag, 2019-07-12:
@joperezr OK. Will look into it. I have just created a Debian installation to play with so I'll probably start with that. As a side issue (and I have only spent 5 minutes on it) but it seems that the I2C devices are not working for whatever reason and I can't get an SPI device to show in /dev/spi*. Having said all that I want to finish off the preliminary work on #559

joperezr, 2019-07-12:
It does work, you only need to make sure you enable both Spi and I2c protocols. In Raspbian doing this is very straight forward as you can just do `sudo raspi-config` and do it from there, but Debian doesn't have such an easy access. I just did this for I2c, so I know that all you have to do is to add 
```
i2c-bcm2708
i2c-dev
```

into the `/etc/modules` file. Spi should be very similar and I have done it in the past but I forgot the name of the modules, I'm sure doing a web search on this shouldn't be too hard to find.

Frankenslag, 2019-07-12:
@joperezr Yes I got it going just before you replied but it,s i2c-bcm2835 I believe. Spi seems different. I wonder if it’s because spidev is compiled into the kernel rather than being a loadable module. Will see if I can work it out.

Frankenslag, 2019-07-13:
@joperezr I think I have worked out the SPI issues. It looks like the Debian image contains a very basic device tree that doesn't respond to dtparam=spi=on.

The solution, from [here](https://github.com/chschlue/rpi3-dt-overlays) creates the following overlay and it can be enabled with `dtoverlay=spi`

```
// Enable spi
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2710";

	fragment@0 {
		target-path = "/soc/spi@7e204000";
		__overlay__ {
			status = "okay";
			spi-max-frequency = <2000000>;
			spidev@0 {
				compatible = "spidev";
				spi-max-frequency = <2000000>;
				reg = <0>;
			};
		};
	};

	fragment@1 {
		target-path = "/soc/spi@7e215080";
		__overlay__ {
			status = "okay";
			spi-max-frequency = <2000000>;
			spidev@0 {
				compatible = "spidev";
				spi-max-frequency = <2000000>;
				reg = <0>;
			};
		};
	};

	fragment@2 {
		target-path = "/soc/spi@7e2150c0";
		__overlay__ {
			status = "okay";
			spi-max-frequency = <2000000>;
			spidev@0 {
				compatible = "spidev";
				spi-max-frequency = <2000000>;
				reg = <0>;
			};
		};
	};
};
```

Frankenslag, 2019-07-14:
I have submmited PR #595

Ellerbach, 2021-07-08:
[Triage] Closing as addressed by the PR #595

closed 2021-07-08 by Ellerbach