dotnet/iot #581: Add Polarity to PwmChannel

shaggygi, 2019-07-09:
**NOTE**: This issue was created before the new `PwmChannel` was added to API, but created as a placeholder for possible future feature.   The initial release has limitations where it only supports active high (idles low) pulses.  Therefore, you will need extra hardware (inverters, transistors, etc.) to convert signals if circuitry requires active low (idles high) pulses.

More details may be added later, but use the links below as reference for now.
- https://github.com/dotnet/iot/pull/576#discussion_r301008653
- https://github.com/dotnet/iot/pull/576#discussion_r300865292

Ellerbach, 2020-12-22:
@shaggygi I feel this has been addressed in various PR, can you confirm and then we can close this issue?

shaggygi, 2020-12-22:
@Ellerbach No, polarity feature has not been added.

krwq, 2020-12-22:
@Ellerbach no, I pushed back on this because negated polarity is equivalent to (1.0 - DutyCycle) which IMO would make it an utility (it's possible to add this functionality by creating a wrapper on top of PwmChannel):

```csharp
public ExtendedPwmChannel : PwmChannel
{
  private PwmChannel _inner;
  public ExtendedPwmChannel(PwmChannel inner) { _inner = inner; }
  public bool ReversedPolarity { /* when setting invert duty cycle if needed */ }
  // .. override everything and passthrough except DutyCycle
  public override double DutyCycle { /* invert logic heere when polarity is reversed*/ }
}
```

shaggygi, 2020-12-22:
It's been many nights slept since we originally discussed, so I can't remember all the thinking behind this.  While I agree on parts of what @krwq is proposing, this issue is not really about the Duty Cycle.  It is more of the state the pin is in when the modulation is not active.  This could cause issues with hardware where it would be driving circuitry (LEDs, motors, etc.) when not active in some cases.  It would force the hardware to only work in active-high states and, in most-cases, hardware is more power-efficient on sinking rather than sourcing.

Please review my previous comment links that explain in more detail.  Thx

krwq, 2020-12-23:
Fair point, we should investigate if the hardware PWM actually supports that before we proceed in that case but also make sure that default implementation will emulate this with software the way I described it

Ellerbach, 2021-07-08:
[Triage] @krwq I remember you've been working on PWM at some point, has this been addressed?

krwq, 2021-07-09:
IIRC this haven't been addressed. Honestly unless we have hardware which explicitly supports this option I'm against this. For @shaggygi's comment above there are already Start/Stop which make hardware do nothing on inactivity (if I understood that correctly)

shaggygi, 2021-07-09:
> For @shaggygi's comment above there are already Start/Stop which make hardware do nothing on inactivity (if I understood that correctly)

This is a main reason why polarity is needed (don't think of pulses at this point).  When inactive, it could be in a state that activates the circuitry that it is driving.  Use driving a motor and tied to Vcc for example (and mentioned previously).  If the inactive state was logic low, the motor would be turned on constantly.  Same for a motor tied tied to GND and the inactive state was logic high.

Need the ability to set the polarity as various hardware could be either case (active low or high).

krwq, 2025-02-06:
[Triage] We really only use this for servo and LED brightness or similar. We recommend to use those abstractions directly rather than extending on PWM. There haven't been any further requests for this so closing. Please re-open if you disagree.

closed 2025-02-06 by krwq