dotnet/iot #583: SPI wait time before sending the data

Ellerbach, 2019-07-09:
I tried to implement an RFID/NFC reader thru SPI: PN532.
Documentation of the device is here: https://www.nxp.com/docs/en/user-guide/141520.pdf
Page 61 describes the mechanism to wake up the device. In short, the line NSS (SPI select pin) has to be polled down for couple of milliseconds before sending data.
Today, with the current implementation, clock is trigged right after NSS state is changing and it's not possible to put a delay.
If you are using another GPIO instead of the built in ones, you can make it manually and force the behavior like this:
```csharp
_controller.Write(pin, PinValue.Low);
Thread.Sleep(1);
_spiDevice.Write(buff);
_controller.Write(pin, PinValue.High);
```
This solution consumes 1 extra GPIO and every time there is a read or write operation, 2 extras steps to change the value of the GPIO needs to be done. 
And it looks like that there is the possibility in the Linux kernel to have this delay built in before sending any data. And that can be setup most likely before all kind of sent.
In case of this sensor, the delay is necessary at every start of a message to ne sent. But it is not necessary for reading.

joperezr, 2019-07-10:
Not sure what is the desired solution here. Whould you instead want a _spiDevice.Write method that takes in a TimeSpan for a delay? You would still need to have the Spi select pin in any case right? I guess I don't really understand what the problem with the existing API is here, but if I understood correctly, then I would expect for this very specific binding to take in a Spi select pin in the constructor so that it can perform the exact thing that you are suggesting above.

krwq, 2019-07-11:
@joperezr I believe this is something which can be added in settings and I think linux driver might be already supporting this.

I think this should be addressed as part of 3.0 since this is something which prevents implementing bindings for certain devices like PN532 (unless we use manual chip select hack as described by @Ellerbach)

Ellerbach, 2021-07-08:
[Triage] Should still be part of the roadmap as should be adressable by the Linux driver and can be with a simple condition in the Windows one. In the mean time, the hack worked as we do have the PN532 already.

closed 2022-06-10 by msftbot[bot]