dotnet/iot #590: System.ArgumentException: No I2C device exists for bus ID 1.

xuliujian2019, 2019-07-12:
I wanted to use GrovePi to connect I2C bus, but it throws an exception as following:

```
Content root path: C:\LED\publish
Now listening on: http://0.0.0.0:9090
Application started. Press Ctrl+C to shut down.
warn: Microsoft.AspNetCore.HttpsPolicy.HttpsRedirectionMiddleware[3]
      Failed to determine the https port for redirect.
fail: Microsoft.AspNetCore.Diagnostics.ExceptionHandlerMiddleware[1]
      An unhandled exception has occurred while executing the request.
System.ArgumentException: No I2C device exists for bus ID 1.
Parameter name: settings.BusId
   at System.Device.I2c.Windows10I2cDevice..ctor(I2cConnectionSettings settings)
   at System.Device.I2c.I2cDevice.Create(I2cConnectionSettings settings)
   at LEDBroker.Controllers.HomeController..ctor() in D:\iot-master\src\devices\LEDBroker\Controllers\HomeController.cs:line 25
   at lambda_method(Closure , IServiceProvider , Object[] )
   at Microsoft.Extensions.DependencyInjection.ObjectFactory.Invoke(IServiceProvider serviceProvider, Object[] arguments)
   at Microsoft.AspNetCore.Mvc.Controllers.ControllerActivatorProvider.<>c__DisplayClass4_0.<CreateActivator>b__0(ControllerContext controllerContext)
   at Microsoft.AspNetCore.Mvc.Controllers.ControllerFactoryProvider.<>c__DisplayClass5_0.<CreateControllerFactory>g__CreateController|0(ControllerContext controllerContext)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.Next(State& next, Scope& scope, Object& state, Boolean& isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.InvokeInnerFilterAsync()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeNextResourceFilter>g__Awaited|24_0(ResourceInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.Rethrow(ResourceExecutedContextSealed context)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.Next(State& next, Scope& scope, Object& state, Boolean& isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.InvokeFilterPipelineAsync()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeAsync>g__Awaited|17_0(ResourceInvoker invoker, Task task, IDisposable scope)
   at Microsoft.AspNetCore.Builder.RouterMiddleware.Invoke(HttpContext httpContext)
   at Microsoft.AspNetCore.Diagnostics.ExceptionHandlerMiddleware.<Invoke>g__Awaited|6_0(ExceptionHandlerMiddleware middleware, HttpContext context, Task task)
fail: Microsoft.AspNetCore.Diagnostics.ExceptionHandlerMiddleware[3]
      An exception was thrown attempting to execute the error handler.
```

My Code:

```
                i2CConnectionSettings = new I2cConnectionSettings(1, GrovePi.DefaultI2cAddress);

                grovePi = new GrovePi(I2cDevice.Create(i2CConnectionSettings));
                led = new Led(grovePi, GrovePort.DigitalPin2);

                grovePi.PinMode(GrovePort.DigitalPin2, PinMode.Input);
```

Environment:
Board: Raspberry Pi 3
Windows IoT Core:Build 17763.503
.Net Core SDK: 3.0.100-preview6-012264

Ogieeeeee, 2019-07-12:
I'm having the same issue here: 

https://stackoverflow.com/questions/56991579/how-to-fix-no-i2c-device-exists-for-bus-id1-grovepi-net-core

krwq, 2019-07-12:
@xuliujian2019 I think we only tested that on Raspbian. Presumably Windows sees Grove Pi on different bus. @joperezr know some way to enumerate I2Cs on Windows IoT Core?

joperezr, 2019-07-12:
We definitely have tested this on Windows in several boards, but I'm not sure if GroovePi was one of them. I believe WinRT Apis do have the capability of looking for available devices on the bus which is similar to what we call so writing a c# app to make the discovery should be possible:

https://github.com/dotnet/iot/blob/1b4efbcb677eea9a07d70216669409ae08f82d38/src/System.Device.Gpio/System/Device/I2c/Devices/Windows10I2cDevice.Windows.cs#L25-L30

Ellerbach, 2020-12-22:
@xuliujian2019 do you still have an issue?
And please check that I2C is correctly activated and you are using the right bus ID as well.

Ellerbach, 2021-07-08:
[Triage] Closing this issue as it seems to be addressed in the comments and how to detect it on Windows. Feel free to reopen in case of anything new.

closed 2021-07-08 by Ellerbach