dotnet/iot #592: SPI mode0, Dataflow LbsFirst creates an exception

Ellerbach, 2019-07-12:
I'm trying to create a SPI device with the Following settings:
```csharp
SpiConnectionSettings settings = new SpiConnectionSettings(0, 0)
{
                ClockFrequency = 2_000_000,
                Mode = SpiMode.Mode0,
                ChipSelectLineActiveState = PinValue.Low,
                DataFlow = DataFlow.LsbFirst
};
UnixSpiDevice device = new UnixSpiDevice(settings);
```

This creates an exception: 
```
Unhandled Exception: System.IO.IOException: Error 22. Can not set SPI mode to Mode0.   at System.Device.Spi.Drivers.UnixSpiDevice.Initialize()
   at System.Device.Spi.Drivers.UnixSpiDevice.Write(ReadOnlySpan`1 buffer)
```

I'm using a RPI 3 with latest Raspbian on it. In this combination, removing the DataFlow will not raise an issue. My sensor, a PN532 required Mode0, Active State Low and LSB First

krwq, 2019-07-12:
As we talked offline, per https://www.kernel.org/doc/Documentation/spi/spidev

```
Several ioctl() requests let your driver read or override the device's current
settings for data transfer parameters:
    SPI_IOC_RD_MODE, SPI_IOC_WR_MODE ... pass a pointer to a byte which will
	return (RD) or assign (WR) the SPI transfer mode.  Use the constants
	SPI_MODE_0..SPI_MODE_3; or if you prefer you can combine SPI_CPOL
	(clock polarity, idle high iff this is set) or SPI_CPHA (clock phase,
	sample on trailing edge iff this is set) flags.
	Note that this request is limited to SPI mode flags that fit in a
	single byte.
    SPI_IOC_RD_MODE32, SPI_IOC_WR_MODE32 ... pass a pointer to a uin32_t
	which will return (RD) or assign (WR) the full SPI transfer mode,
	not limited to the bits that fit in one byte.
    SPI_IOC_RD_LSB_FIRST, SPI_IOC_WR_LSB_FIRST ... pass a pointer to a byte
	which will return (RD) or assign (WR) the bit justification used to
	transfer SPI words.  Zero indicates MSB-first; other values indicate
	the less common LSB-first encoding.  In both cases the specified value
	is right-justified in each word, so that unused (TX) or undefined (RX)
	bits are in the MSBs.
    SPI_IOC_RD_BITS_PER_WORD, SPI_IOC_WR_BITS_PER_WORD ... pass a pointer to
	a byte which will return (RD) or assign (WR) the number of bits in
	each SPI transfer word.  The value zero signifies eight bits.
    SPI_IOC_RD_MAX_SPEED_HZ, SPI_IOC_WR_MAX_SPEED_HZ ... pass a pointer to a
	u32 which will return (RD) or assign (WR) the maximum SPI transfer
	speed, in Hz.  The controller can't necessarily assign that specific
	clock speed.
```

very likely https://github.com/dotnet/iot/blob/76276a269bbf6c47e5f4a547b0937e15a6419f89/src/System.Device.Gpio/System/Device/Spi/Devices/UnixSpiDevice.Linux.cs#L66 will need to be split into two calls (first one mode specific, the other one LSB specific)

krwq, 2019-07-16:
@Ellerbach could you share some tiny repro of this I can play with? I already got my PN-532/

joperezr, 2019-07-18:
@krwq have you been able to repro this? It would be good to get a fix in as soon as possible.

krwq, 2019-07-18:
Haven't tried yet. It's next on my radar after moving SPI stuff out of Mcp3008

krwq, 2019-07-26:
@Ellerbach I'm having issue with the original code. I've only did minimal changes to your PR (only stuff required to make it build + changing UART to SPI):

```csharp
            var settings = new SpiConnectionSettings(0, 0)
            {
                ClockFrequency = 2_000_000,
                Mode = SpiMode.Mode0,
                ChipSelectLineActiveState = PinValue.Low,
                //DataFlow = DataFlow.LsbFirst
            };
            using (var spi = SpiDevice.Create(settings))
            using (var pn532 = new Pn532(spi))
...
```

I wanted to make sure original code is working first before I keep going. For SPI I connected NSS/CS to pin 18 as code suggested.

I'm seeing the issue above when I uncomment Lsb part but I don't feel confident sending a patch without trying out if it actually fixes the issue

joperezr, 2019-07-29:
Chatted with @krwq offline. Given that for now there is only one device affected by this I think fixing the issue is lower priority for now, so I will mark it to future so that we can focus on other issues first. If we do eventually find a fix for this, we can still fix it in time before release, the idea of moving it to future is simply to not block the release on this issue.

Ellerbach, 2019-07-31:
> Chatted with @krwq offline. Given that for now there is only one device affected by this I think fixing the issue is lower priority for now, so I will mark it to future so that we can focus on other [...]

I'm fine with that. I manage to have an experimental mode for SPI and the device provide 2 other ways to communicate thru High Speed UART and I2C.

danieljloughlin, 2020-12-27:
Just to say this if frustratingly still an issue in System.Device.Gpio 1.3.0

Ellerbach, 2020-12-29:
> Just to say this if frustratingly still an issue in System.Device.Gpio 1.3.0

The issue is with the Raspberry Pi kernel driver not supporting this mode. This will have to be supported as an emulation like in the PN532. We will try to make it transparent but as there is a solution like in the PN532, it's not in the core priority. 
See the pattern with the `ReverseByte` function: https://github.com/dotnet/iot/blob/65ce9658b69336f9c39d0b24e04293b15f31360c/src/devices/Pn532/Pn532.cs#L1827
You can open the SPI port in MSB and use the Reverse function when you write and read thru SPI. Copy the functions and the static, you'll be good to go.

Ellerbach, 2021-02-18:
[Triage] Duplicate now with #1308 and PN532 implement this.

closed 2021-02-18 by Ellerbach