dotnet/iot #601: SpiConnectionSettings do not work well with software SPI implementation

krwq, 2019-07-18:
Current settings don't work very well with future SoftwareSpi implementation since BusId and ChipSelectLine are meaningless there.

Example solution 1:
- SpiConnectionSettings should not contain BusId and ChipSelectLine
- SpiDevice.Create would have 2 overloads:
  - BusId/ChipSelectLine
  - Pin

Example solution 2:
- Add another constructor and property which takes Pin, internally we can reuse one of the int-s for the second purpose and just make it look nice from the outside with properties

joperezr, 2019-07-18:
perhaps the SpiDevice.Create() for SoftwareSPI shouldn't use the SpiConnectionSettings and instead use a different mechanism of selecting the SpiMode, data big length, etc?

krwq, 2019-07-18:
I feel like SpiConnectionSettings is not useful at all, we should just move common options to SpiDevice and whatever is hardware/software specific move it to that class

shaggygi, 2019-07-21:
@krwq  I see your point that BusId doesn't seem to fit with the software version, but I like being able to use setting parameters to define the setup similar to native SpiDevice.  I honestly don't think the software device will be used much (even though I like having the option).  So I guess the question would be... does it matter if you are required to define a BusId, even if it is something like -1 most of the time?

I can see where users can get confused on ChipSelectLine as it is 0/1 for native SpiDevice and not the actual pin number.  However, I believe ChipSelectLine should be used as the pin number when using the software version.  So instead of having another parameter named "cs", ChipSelectLine would be the pin number.

```csharp
// Define with cs parameter, but added to settings within.
public SoftwareSpi(int clk, int miso, int mosi, int cs, GpioController controller = null)
{
    var settings = new SpiConnectionSettings(-1, cs);
    // other init code...
}

// and offer settings in overload where chip select (and other parameter options) are defined.
public SoftwareSpi(SpiConnectionSettings settings, int clk, int miso, int mosi, GpioController controller = null)
{
    // init code...
}
```

I'm still not convinced that one day there will be a need for a SpiController.  While it's no biggie, I'm having to implement my own in a personal project since we decided not to include in core API.  BusId and ChipSelectLine would most likely be the parameters used for the unique identifiers.  Meaning, you could have multiple devices on bus X to use the common data/clock pins, but each would have to use a unique chip select pin.

krwq, 2019-08-14:
I don't believe we will address this issue in 3.0 timeline and therefore we will leave this as is.

closed 2019-08-14 by krwq