dotnet/iot #602: SpiConnectionSettings should be calle SpiSettings

krwq, 2019-07-18:
Some of the Spi options feel like they don't have anything to do with how the connection is established and rather with how transmission is defined and therefore the `Connection` part of the name does not only make the name longer but also confusing

joperezr, 2019-07-18:
WinRT uses the exact same name and contains the same info: https://docs.microsoft.com/en-us/uwp/api/windows.devices.spi.spiconnectionsettings 

I wouldn't really want to change this at this point unless there is a very important reason why to do it.

shaggygi, 2019-07-21:
I'm on the fence and agree with both.  Just a FYI... there is already an internal [SpiSettings](https://github.com/dotnet/iot/blob/4fa64a1b1469f87b0fc86412e7b2cf431c1b581b/src/System.Device.Gpio/Interop/Unix/Libc/Interop.ioctl.cs#L79), but shouldn't cause any issues.  I guess now or never if this will be changed.

Ellerbach, 2021-07-08:
[Triage] Now we've been using this naming for a long time, we will just keep it. That would introduce some braking changes. Closing this issue, if anything new is coming and everyone would prefer a braking change, please reopen.

closed 2021-07-08 by Ellerbach