dotnet/iot #603: SpiConnectionSettings should be immutable or cloned by SpiDevice

krwq, 2019-07-18:
Currently properties in SpiConnectionSettings have setters which makes them mutable. Also SpiDevice takes settings and simply assigns them. This has side effect that changing settings after SpiDevice is constructed may have effect on SPI transmission

Frankenslag, 2019-07-18:
I think I would prefer the clone as I think that having the individual properties is quite useful.

joperezr, 2019-07-18:
I think this is a valid concern and an easy enough fix. So the idea would be to make properties `{ get; private set; }` right? @krwq are you planning on putting a PR for this?

Frankenslag, 2019-07-19:
@krwq @joperezr 

I like using object initializers i.e.

`mySettings = new SpiConnectionSettings(0, 0) { Mode = SpiMode.Mode3, DataBitLength = 8, ClockFrequency = 12_000_000 /* 12MHz */ }`

If the properties are` { get; private set; }` won't that mean that we will have to come up with a whole bunch of constructors and then everytime there is a new then extend the constructors ?

This is why I prefer the clone option within the device as well as cloning meaning that existing code continues to work.

krwq, 2019-07-19:
I think the best solution would be to get rid of SpiConnectionSettings completely and just put those directly on SpiDevice - they don't seem to be working very well with software implementation and also make problems like the one here

Frankenslag, 2019-07-19:
@krwq 

Again, not sure about this because we would be swapping an immutable class with a number of immutable properties on the SpiDevice which seems harder.

I can see that a number of areas of the framework use settings classes but I am not aware of why. Perhaps it’s because having all the configurable items in one class makes it trivial to serialize to a config file.

Anyway just my 2 cents.

krwq, 2019-07-19:
@Frankenslag but then we have stuff like ChipSelectLine/BusId on the settings which seems very weird considering it doesn't mean anything for software SPI implementation (except for maybe ChipSelectLine) and that one also have explicit MISO/MOSI/CLK pins which also wouldn't serialize.

I need to think about it a bit more and see if I can make the abstraction better

shaggygi, 2019-07-21:
@krwq I see your point, but I also like the object initializers.  I'm wondering if we could just pass in all parameters as optionals with default values.  I believe we now have all the settings needed for most use-cases (native, software, etc.), but this could be breaking change if we added a future setting.

PR #656 (2019-08-05): Clone I2cConnectionSettings and SpiConnectionSettings in order to make them immutable
fixes #603 

cc: @shaggygi @Frankenslag @krwq 

These changes will fix the current issue where you can modify the connection settings of a device after creation. It would still allow people to change settings object themselves in order to allow dynamic instantiation (see comment https://github.com/dotnet/iot/issues/603#issuecomment-513172714) and when the settings are requested form the device, it will just create a clone and pass it to the caller.
closed 2019-08-05 by joperezr