dotnet/iot #616: Preparing for Nullable Reference Types

shaggygi, 2019-07-22:
I was looking into preparing the core API to use the C# 8 Nullable Reference Types.  Based on chat with @joperezr, it would be better to add "#nullable enable" to individual files instead of adding to csproj while progress is made.  I'm assuming once all files are updated/reviewed, the "#nullable enable" could be removed and <Nullable>enable</Nullable> added to csproj.

One question I had is related to disposing.  Use [Windows10SpiDevice.Windows.cs](https://github.com/dotnet/iot/blob/030958af2d2ac42b153cde403cc8e219501accc7/src/System.Device.Gpio/System/Device/Spi/Devices/Windows10SpiDevice.Windows.cs#L121) as example.

```csharp
// Currently...
protected override void Dispose(bool disposing)
{
    _winDevice?.Dispose();
    _winDevice = null;

    base.Dispose(disposing);
}
```

Since _winDevice should not be nullable, should it be changed to...

```csharp
protected override void Dispose(bool disposing)
{
    _winDevice.Dispose();
    base.Dispose(disposing);
}
```

joperezr, 2019-07-22:
First of all I don't really think that this will be easy for us to do now, since only a very small subset of the framework has been annotated so it is likely that we might encounter parts where the framework can still be returning null and we don't catch it, hence our consumers won't either. Also, we target .NET Standard which I'm not quite sure if it'll ever be annotated (cc: @safern in case he knows). As to the above question, I think that we still want to mark _winDevice as null so that we won't be calling Dispose on that twice, in case we do end up getting called twice for disposing.

shaggygi, 2019-08-07:
Referencing [Try out Nullable Reference Types](https://devblogs.microsoft.com/dotnet/try-out-nullable-reference-types/)

safern, 2019-08-07:
> Also, we target .NET Standard which I'm not quite sure if it'll ever be annotated

We don't know yet. It might be annotated once we're done with the whole framework in the 5.0 timeframe (we don't know if that's the goal yet), however, currently we've only annotated 20% of the framework.

shaggygi, 2019-12-02:
Referencing... https://devblogs.microsoft.com/dotnet/embracing-nullable-reference-types/

safern, 2019-12-02:
We also have these guidelines for .NET Core libraries: https://github.com/dotnet/runtime/blob/master/docs/coding-guidelines/api-guidelines/nullability.md

Plus here is our tracking issue to annotate all of the shared framework: https://github.com/dotnet/corefx/issues/40623

krwq, 2019-12-02:
I think we should wait with this work until framework is done annotating but I'm fine with merging if anyone is willing to contribute earlier since the major stuff affecting us is mostly done already

Ellerbach, 2020-12-22:
All nullable work has been done with moving to .NET 5.0. Co closing this issue.

closed 2020-12-22 by Ellerbach