dotnet/iot #617: Should devices folder be capitalized

shaggygi, 2019-07-22:
Not really a biggie, but should the [devices folder](https://github.com/dotnet/iot/tree/master/src/devices) by capitalized to be consistent with others?

[image]

krwq, 2019-07-22:
I'd refrain from doing that at least until we ship 3.0

joperezr, 2019-07-22:
I don't think we should do this. The other folders are capitalized only because they have the libraries names, which are capitalized. We would have to change many other folders if we wanted to do this, like "src" and "tools" or "samples". Making all of these changes might have unintended consequences on other OSes and it also doesn't echo the model we have followed in corefx and the rest of our repos.

shaggygi, 2019-07-22:
> because they have the libraries names

Yeah, that's what I was thinking... but wasn't completely sure.  I'd say leave as in since src, tools and such are similar.  Close if no further work is needed.  Thx

closed 2019-07-22 by joperezr