dotnet/iot #627: Development work on Windows 10

alex182, 2019-07-25:
I am attempting to do development work on a Windows 10 machine for a project that will be used on a Raspberry Pi.  The only issue I am running into is that when I attempt to run my project on the Windows 10 machine I get the following exception `No GPIO controllers exist on this system.`.

Now I understand why that is happening (my laptop obviously doesn't have GPIO pins) but I was wondering if there was a built in way to suppress this error?

krwq, 2019-07-25:
@alex182 do you mean i.e.:

```csharp
using (var controller = new GpioController())
{
}
```

? What would you do with such instace? Could you give some usage example?

Frankenslag, 2019-07-25:
I am not really answering your question, I know, but I am not sure why you want to execute on the Win10 machine. I have found it reasonably simple to publish the resultant code to a RaspberryPi and then debug using a combination of VS2019 and vsdbg on the Pi.

This allows you to develop and debug using familiar tools if you are a Windows developer but also see the effects of manipulating Gpio, Spi, I2c etc.

alex182, 2019-07-25:
@krwq I am developing an api that runs on the Pi but there are non-GPIO things that I want to be able to debug on my Windows 10 machine (logging, validation etc)

joperezr, 2019-07-25:
@alex182 Does your app do other things apart from just controlling the microcontroller? I suppose it does and then that is why you want to just mock that part so you can continue debugging the rest of your code. I'm not sure we have a good way of doing that, but I would suggest two things:

1. You could locally create a class that extends the [GpioDriver](https://github.com/dotnet/iot/blob/master/src/System.Device.Gpio/System/Device/Gpio/GpioDriver.cs) which simply doesn't do anything, and then use that to initialize your GpioController. So adding a class that looks like this:
```c#
class GpioDriverMock : GpioDriver
{
  protected override int PinCount => 0;

  protected override void AddCallbackForPinValueChangedEvent(int pinNumber, PinEventTypes eventTypes, PinChangeEventHandler callback) { }

  protected override void ClosePin(int pinNumber) { }

  protected override int ConvertPinNumberToLogicalNumberingScheme(int pinNumber) => 0;

  protected override PinMode GetPinMode(int pinNumber) => default(PinMode);

  protected override bool IsPinModeSupported(int pinNumber, PinMode mode) => true;

  protected override void OpenPin(int pinNumber) { }

  protected override PinValue Read(int pinNumber) => default(PinValue);

  protected override void RemoveCallbackForPinValueChangedEvent(int pinNumber, PinChangeEventHandler callback) { }

  protected override void SetPinMode(int pinNumber, PinMode mode) { }

  protected override WaitForEventResult WaitForEvent(int pinNumber, PinEventTypes eventTypes, CancellationToken cancellationToken) => default(WaitForEventResult);

  protected override void Write(int pinNumber, PinValue value) { }
}
```

And then you could just add this to use this mock when creating your controller:

```c#
using (GpioController controller = new GpioController(PinNumberingScheme.Logical, new GpioDriverMock()))
{
  // Do whatever with your controller.
}
```

Once you are ready to test in in actual hardware, simply remove the  parameters passed in to `new GpioController()`.

2. The other option would be to just add a try catch to the places where you are hitting the exception, and simply remove them whenever you are ready to deploy.

alex182, 2019-07-25:
@joperezr thats exactly what I was looking for.   And yes the app does more than accessing the micro controller.  Thank you.

joperezr, 2019-07-25:
Cool, closing for now, feel free to reopen if you need any additional help.

closed 2019-07-25 by joperezr