dotnet/iot #629: AddCallbackRemoveAllCallbackTest is failing intermitently in test runs

joperezr, 2019-07-25:
Because of the way that we currently have our SysFs Gpio Driver implemented, this test may run into a race condition intermittently and cause it to fail. Because of that, I'm going to disable it for now when executing SysFs driver tests or RaspberryPi driver tests as they use the same logic underneath. This issue is to track the work to make eventing logic work better for those drivers, and to enable the test once that is done.

krwq, 2020-09-23:
It seems to be flaky also on Windows so I'm disabling this comepletely

Ellerbach, 2021-07-15:
[Triage] This is still valid. SysFs driver is used only in few boards not supporting anything else. What should be added is a warning if libgpio is not installed maybe?

krwq, 2025-02-06:
[Triage] This seems to have been fixed at one point

closed 2025-02-06 by krwq