dotnet/iot #632: RTC devices not setup for device-listing

shaggygi, 2019-07-26:
I haven't looked in any detail, but it seems like the recent changes/addition to [Rtc binding folder](https://github.com/dotnet/iot/tree/master/src/devices/Rtc) doesn't work with [device-listing](https://github.com/dotnet/iot/tree/master/tools/device-listing) generation now.

krwq, 2019-07-26:
let me re-gen and see what's going on

krwq, 2019-07-26:
@shaggygi device-listing is not being auto-generated - you have to manually run the tool (I've found manual better so far since it gives me a chance to review the changes). When I run `dotnet run` on device-listing dir I'm seeing:

```
Warning: /home/krwq/work/iot/src/devices/SoftwareSpi/category.txt is missing
Warning: /home/krwq/work/iot/src/devices/Rtc/category.txt is missing
```

so that seems to be a culprit - I'll fix it and send PR with the changes

closed 2019-07-26 by krwq

shaggygi, 2019-07-26:
Yeah, i noticed that, which is a nifty utility.  It seemed like it wasn’t generating for Clocks section.