dotnet/iot #637: System.Device.Gpio need Abstractions

Ordjony, 2019-07-29:
Propose to split **System.Device.Gpio** into different libs like **System.Device.Gpio.Abstractions** (e.g. asp.net core) without dependencies and **System.Device.Gpio.*** (OS or device-specific).

At the moment System.Device.Gpio has dependencies on Microsoft.Win32.Registry, System.Runtime.InteropServices.WindowsRuntime, System.Runtime.WindowsRuntime which looks like not necessary on Linux. Iot.Device.Bindings should use Abstractions library.

This Abstractions package gives the ability to use binding in different platforms (e.g. https://www.wildernesslabs.co/meadow).

joperezr, 2019-07-29:
Hey @Ordjony thanks a lot for logging the issue. Is the only concern that there are extra Windows-specific dependencies on the library package? If so, I don't think that this would warrant the fact of creating a new abstraction layer. All those three dependecies you found are only really used by the windows implementation, none of those dependecies are exposed in the API shape, and of course they aren't used at all in the Linux implementation. Also, those dependencies have packages that are supported in netstandard, so there is really no concern that the packages won't restore correctly on other non-Windows platforms. There is no good way in NuGet to add dependencies to a package that are specific only to one RID, which is why we have to add it to all of them but that doesn't mean that we need those dependencies everywhere. This should in no way limit where would IoT.Device.Bindings could run, as System.Device.Gpio library code should run anywhere where netstandard 2.0 is supported(provided that we are running in a board that does in fact expose GPIO pins)

Ordjony, 2019-07-29:
Hi, @joperezr thanks for the very fast answer. Now I looked at Nuget package inside and understood that it's no problem to target any platform which supports netstandard 2.0. This dependency was confused.

joperezr, 2019-07-29:
Thanks again for your input and suggestion. If there is a specific binding you were interested in let us know so that we can create an issue for it and start taking for PRs.

Ordjony, 2019-09-11:
I decided to reopen my issue and continue the discussion. I researched a little bit ARM MBED OS and watched to Rust lang HAL.
I think in .net we need our own HAL library which will contain only interfaces for GPIO without any methods to get this GPIO or settings.
```csharp
public interface II2CBus
{
    void Read(Span<byte> buffer);
    byte ReadByte();
    void Write(ReadOnlySpan<byte> buffer);
    void WriteByte(byte value);
    void WriteRead(ReadOnlySpan<byte> writeBuffer, Span<byte> readBuffer);
}
```
Current API like I2cConnectionSettings is applied only for Raspberry PI or similar it's not easy to use with MCU. Because we have Linux\Windows OS which abstract 2 pins SDA, SCL to busId. But in MCU we usualy use just two pins number.

joperezr, 2019-09-11:
When you say that it only uses two pins, do you mean it uses a Software implementation of I2C? If so, we can do something similar to what we did for the [software implementation of SPI](https://github.com/dotnet/iot/blob/master/src/devices/SoftwareSpi/README.md) where we basically have software SPI extend from SPIDevice so you can initialize it using the provided pins. We generally try to avoid using Interfaces as abstractions because they don't version well (you can't add members after you ship them) which is why we instead used I2cDevice and SpiDevice as the abstractions which are abstract classes (and that means we can add members in the future and just provide a default implementation)

krwq, 2019-09-11:
I2cDevice already looks very similar to what you showed above. Unfortunately until we can use C# 8 (hopefully next/5.0 release) we don't really have a good versioning story for interfaces.

What is the scenario where you need interface vs just abstract class?

Ordjony, 2019-09-12:
An abstract class is ok. My idea is to make HAL library which will contain only abstractions for I2C, SPI, PWM, GPIO. In my opinion, .NET needs this HAL library for standardization all IoT\MCU interfaces (like do .net core team with logging.abstraction, DependencyInjection.Abstractions libraries).
I found at the moment two projects which are running .net on MCU:
1. Meadow.NET - own RTOS which can runs .net standard 2.0 apps on Mono.
2. TinyCLR - own TinyCLR OS which based  .NET Micro Framework.
The problem here's they both produce their own API for GPIO and own libraries for access to some hardware (LCD, Motors etc.).

I think if Microsoft creates HAL library other projects possibly can use it and it's win-win for all. Developers can share their drivers\libs for different hardware, and anybody can use it does not matter what RTOS\Linux\Windows they use.

Ellerbach, 2021-07-15:
[Triage] Since this issue has been opened, a large work has been done to make all drivers outside of the implementation. We do have board definition as well, I2C Bus and the Gpio, I2C, SPI, etc has been unsealed as well.
We've been doing some API alignment with .NET nanoFramework based on the definition for example and it was very easy to align the API. The implementation is specific of course. So for other platforms which as very different, that should be sufficient.
So, closing this issue. Still if you feel that there are more elements and a need to reopen it, feel free to do so.

closed 2021-07-15 by Ellerbach