dotnet/iot #644: Abnormal data read from BME280

ZhangGaoxing, 2019-07-31:
My BME280 is normal. I tested it with Arduino. But when I use `Iot.Device.Bindings` to read, the temperature is higher and the pressure is lower. Here is my code, I used a timer to read.

```C#
private static async void Timer_Elapsed(object sender, ElapsedEventArgs e)
{
    I2cConnectionSettings settings = new I2cConnectionSettings(0, Bme280.SecondaryI2cAddress);
    I2cDevice device = I2cDevice.Create(settings);

    using Bme280 bme = new Bme280(device);
    using WeatherContext context = new WeatherContext();

    bme.SetPowerMode(Bmx280PowerMode.Forced);

    double t = Math.Round((await bme.ReadTemperatureAsync()).Celsius, 2);
    double h = Math.Round(await bme.ReadHumidityAsync(), 2);
    double p = Math.Round(await bme.ReadPressureAsync(), 2);

    Console.WriteLine($"Temperature:{t} Humidity:{h} Pressure:{p}");

    context.Add(new Weather
    {
        DateTime = DateTime.Now,
        Temperature = t,
        Humidity = h,
        Pressure = p
    });
    context.SaveChanges();
}
```

## Scenario 1
If I remove `bme.SetPowerMode(Bmx280PowerMode.Forced)`, the temperature looks good, the humidity and the perssure are abnormal(The humidity is about 39%, and the pressure is about 100,000Pa.), and the data has never changed.
[image]

## Scenario 2
If I don't remove `bme.SetPowerMode(Bmx280PowerMode.Forced)`, the humidity and the pressure are normal, but the temperature is too high.
[image]

## Scenario 3
Exception will occur if the reading interval is too short, such as 1s. I didn't test on the Raspberry Pi. I used an ARM Cortex-A7 CPU board. Maybe it's related to the CPU's speed. This board does run slowly, but it's compatible with iot. It's not a big problem, I record it here first.
[image]

## Suggestions
1. It's better to keep 2 decimal digits in the read data.
2. Why use async methods? I didn't look at the source code of BMPxxx. But I've coded BMP180, and I didn't use async methods.

RobinTTY, 2019-07-31:
While working on adding new functionality for the Bme680 device of the same family, I found a small mistake in the temperature compensation formula of Bmxx80Base: https://github.com/dotnet/iot/pull/482/commits/d4442c859452db8b8bae0f0166966b6990c2b28f

But the problem you describe doesn't sound necessarily related, since it changes based on which power mode is chosen. Maybe I'm able to find the problem once I get around to debugging but I can only test this on a Bme680 which does not have a forced power mode.

ZhangGaoxing, 2019-07-31:
> I get around to debugging but I can only test this on a Bme680 which does not have a forced power mode.

Normal mode also haha🤔

ZhangGaoxing, 2019-07-31:
[image]

The pressure data are also inaccurate. Is calibration data ok? Humidity data should be no problem, and the humidity meter on my desk is the same value. The Arduino library is [Adafruit_BME280_Library](https://github.com/adafruit/Adafruit_BME280_Library)

RobinTTY, 2019-07-31:
This rather small difference could very well be caused by the mistake that is currently in the temperature compensation formula (since the temperature is a component in the humidity and pressure formula).

The mistake is here:

```C#
double var1 = ((adcTemperature / 16384.0) - (_calibrationData.DigT1 / 1024.0)) * calibrationData.DigT2;
double var2 = ((adcTemperature / 131072.0) - (_calibrationData.DigT1 / 8192.0)) * -->_calibrationData.DigT3<--; // this multiplication should only happen once not again in the next line
var2 *= var2 * _calibrationData.DigT3;
```
I would therefore expect the current value to be too high, which matches your results.
I already changed this in PR #482 for the Bme680.

RobinTTY, 2019-07-31:
But the difference in the temperature measurement is rather strange to me 😄

krwq, 2019-07-31:
@RobinTTY Would be cool to merge this PR soon then. What's left in there? Is it an option to extract that fix from your PR and create a separate one just targeting this issue?

RobinTTY, 2019-07-31:
@krwq sure I'll create a PR for it

RobinTTY, 2019-07-31:
@krwq Fixed in #647

ZhangGaoxing, 2019-08-02:
close via #647

closed 2019-08-02 by ZhangGaoxing