dotnet/iot #646: Cannot build for arm32 when use System.Device.Gpio in dotnet core preview 7 Web Application project.

yan0lovesha, 2019-07-31:
I want to use System.Device.Gpio in a Asp.Net core WebAssembly project. I am using dotnet core preview 7. The repro steps are as below:

1. Create a solution in VS preview by using Asp.Net WebApplication Blazor WebAssembly template.
2. Upgrade all nuget package to preview7 version if any is still preview6.
3. In the Server side project, add reference to System.Device.Gpio.
4. Save all. 
5. Run **dotnet publish . -r linux-arm** in the solution folder to build and publish the bits for linux-arm .

Then I see the error like "detected package downgrade from 4.3.0 to 4.0.1"

joperezr, 2019-07-31:
Can you please share a binlog of your project to try to see what is going on? To produce one, please run the following on a developer command prompt: `msbuild yourSolution.sln /t:rebuild /bl:Build.binlog` this will create a Build.binlog file, can you please attach that here so that we can investigate?

krwq, 2019-07-31:
@yan0lovesha please also share `dotnet --info` and what is the version of System.Device.Gpio you are using

yan0lovesha, 2019-08-01:
@joperezr @krwq , I attached 3 files for the result of binlog, dotnetinfo and dotnet_publish.

[dotnet_publish.txt](https://github.com/dotnet/iot/files/3455005/dotnet_publish.txt)
[dotnetinfo.txt](https://github.com/dotnet/iot/files/3455006/dotnetinfo.txt)
[binlog.txt](https://github.com/dotnet/iot/files/3455007/binlog.txt)

yan0lovesha, 2019-08-01:
This is the correct binlog file.
[Build.binlog.zip](https://github.com/dotnet/iot/files/3455022/Build.binlog.zip)

joperezr, 2019-08-01:
The binlog that you provided doesn't have any warnings or errors about package downgrade, and they also don't have System.Device.Gpio as a reference. How are you trying to add the dependency to your project? Can you manually add the reference by editing the .csproj and then do a `dotnet publish . -r linux-arm /bl:Build.binlog` that way we can see exactly the repro of the issue and investigate.

krwq, 2019-08-01:
I think sharing csproj + commands you are using to build and run (also mention OS and architecture or what was run) should help us investigate

yan0lovesha, 2019-08-02:
Sorry, I think I attached a wrong binlog file yesterday. That was generated before I add gpio reference. This one is the one with error message.

[binlog.zip](https://github.com/dotnet/iot/files/3459629/binlog.zip)

joperezr, 2019-08-02:
Ok I know what is going on. This isn't a problem with System.Device.Gpio package, it is actually caused by a [NuGet bug](https://github.com/NuGet/Home/issues/3541) and the fact that we don't yet have a [good silent workaround in the sdk](https://github.com/dotnet/sdk/issues/3044). Even if you remove the System.Device.Gpio package reference and try to rebuild with `-r linux-arm` you will hit the same issue, which would prove that System.Device.Gpio package is not the cause of this. For now, in order to fix this you have to manually add a reference to the new Microsoft.NETCore.Targets package into your project like:
```xml
  <ItemGroup>
    <PackageReference Include="Microsoft.AspNetCore.Blazor.Server" Version="3.0.0-preview7.19365.7" />
    <PackageReference Include="Microsoft.AspNetCore.Mvc.NewtonsoftJson" Version="3.0.0-preview7.19365.7" />
    <PackageReference Include="System.Device.Gpio" Version="0.1.0-prerelease.19376.1" />
    <PackageReference Include="Microsoft.NetCore.Targets" Version="2.1.0" />   <!--  Add this reference to your project -->
  </ItemGroup>
```

That would fix the issue. The two bugs I pointed you at (the nuget and the sdk bug) would need to get fixed in order for you not to need to add manually this package. 
I have validated on my end that this workaround fixes your issue and given that there are already a couple bugs in the right repos to track this issue I'm going to close this one, but feel free to reopen if you think we should keep this open as well as if you need any help in getting your project working. Thanks a lot for reporting this again!

closed 2019-08-02 by joperezr