dotnet/iot #650: Add samples/tutorials on how to connect devices with Azure services

krwq, 2019-08-01:
Currently we have a lot of samples for each device separately but not many showing end to end scenarios. It would be to really good to add some kind of tutorial showing end to end how to put everything together and create a simple service using some of our existing device bindings.

Once tutorial is up we can publicize this repo and link to the tutorial which would cause more people to start using it and potentially start contributing and helping with the effort.

cc: @ericstj

Frankenslag, 2019-08-02:
@krwq 

I am thinking about coming up with an FM radio as a demonstration. I am working on a RotaryEncoder binding for the volume and tuning functions and then a binding for an FM tuner all wrapped up with a display of some sort. I can't think how to involve Azure with this though.

joperezr, 2019-08-02:
I have a demo of a website using Azure IoT Hub to control a couple of light bulbs connected to a Pi using our library, kind of a hello world home automation project so I could add that too.

krwq, 2019-08-02:
@Frankenslag possibly selecting FM frequency on the website and streaming sound?

@joperezr cool, could you add it any time soon?

joperezr, 2019-08-02:
> cool, could you add it any time soon?

sure I can probably work on this next week, just need to find out where I've put it 😆

Ellerbach, 2021-07-15:
[Triage] Some work has been done, still not directly posted on this repo ([showdotnet](https://devblogs.microsoft.com/dotnet/) for example). More to come!
So closing this issue.

closed 2021-07-15 by Ellerbach