dotnet/iot #654: The LibGpiodDriver.Linux cannot produce both rising and falling pin events.

Frankenslag, 2019-08-04:
If I look at AddCallbackForPinValueChangedEvent and WaitForEvent in the LibGpiod driver then I see the following 

```
if ((eventTypes & PinEventTypes.Rising) != 0)
{
      eventHandler.ValueRising += callback;
}
else if ((eventTypes & PinEventTypes.Falling) != 0)
{
      eventHandler.ValueFalling += callback;
}
```

The way I read this is that you could not register for both a rising and falling pin event (you would just get the rising one). Is this intentional ? If not do you want me to create a PR ?

krwq, 2019-08-04:
yes, looks like a bug. You can go ahead and create PR

Frankenslag, 2019-08-05:
@krwq
PR #655 created to resolve this.

joperezr, 2019-08-05:
Fixed by #655

closed 2019-08-05 by joperezr