dotnet/iot #657: Can not read a pin value in the PinValueChangedEvent

ZhangGaoxing, 2019-08-06:
I'm trying to bind an OV camera, and I need to use PinValueChangedEvent. After I initialized the camera and called `Capture()` method, the `GpioController.Read()` throw an exception in the event `VSYNC_ValueChanged`.
> Unhandled exception. System.InvalidOperationException: Can not read from a pin that is not open.
   at System.Device.Gpio.GpioController.Read(Int32 pinNumber)
   at Iot.Device.Ov2640.Ov2640.VSYNC_ValueChanged(Object sender, PinValueChangedEventArgs e) in C:\Users\zhang\source\repos\Ov2640\Ov2640\Ov2640.cs:line 107
   at System.Device.Gpio.Drivers.UnixDriverDevicePin.OnPinValueChanged(PinValueChangedEventArgs args)
   at System.Device.Gpio.Drivers.SysFsDriver.DetectEvents()
   at System.Threading.ThreadHelper.ThreadStart_Context(Object state)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location where exception was thrown ---
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart()

Here is my code
```C#
private GpioController _controller;

public Ov2640(I2cDevice i2cDevice, Ov2640Pin pin, PinNumberingScheme scheme = PinNumberingScheme.Logical)
{
    _pin = pin;
    _i2cDevice = i2cDevice;
    _controller = new GpioController(scheme);

    _controller.OpenPin(_pin.D0, PinMode.Input);
    _controller.OpenPin(_pin.D1, PinMode.Input);
    _controller.OpenPin(_pin.D2, PinMode.Input);
    _controller.OpenPin(_pin.D3, PinMode.Input);
    _controller.OpenPin(_pin.D4, PinMode.Input);
    _controller.OpenPin(_pin.D5, PinMode.Input);
    _controller.OpenPin(_pin.D6, PinMode.Input);
    _controller.OpenPin(_pin.D7, PinMode.Input);

    _controller.OpenPin(_pin.PCLK, PinMode.Input);
    _controller.OpenPin(_pin.PWDN, PinMode.Output);
    _controller.OpenPin(_pin.VSYNC, PinMode.Input);
    _controller.OpenPin(_pin.HREF, PinMode.Input);
    _controller.OpenPin(_pin.RST, PinMode.Output);

    Initialize();
}

public void Capture()
{
    _controller.RegisterCallbackForPinValueChangedEvent(_pin.VSYNC, PinEventTypes.Rising, VSYNC_ValueChanged);
    .......
}

private void VSYNC_ValueChanged(object sender, PinValueChangedEventArgs e)
{
    while (_controller.Read(_pin.HREF) == PinValue.Low) ;    // throw an exception
    .........
}
```

joperezr, 2019-08-06:
Thanks for reporting this @ZhangGaoxing. Are you able to debug the program when the exception happens? I would be interested to see if you can debug into the call `_controller.Read(_pin.HREF)` and check what is the content of this hashset: https://github.com/dotnet/iot/blob/master/src/System.Device.Gpio/System/Device/Gpio/GpioController.cs#L17 and checking if _pin.HREF is present.

ZhangGaoxing, 2019-08-10:
@joperezr I printed out `_openPins`, HREF is in the `_openPins`, but the exception is still thrown.

[image]

```C#
public void Capture()
{
    foreach (var item in _controller._openPins)
    {
        Console.Write($"{item} ");
    }
    Console.WriteLine("-------");

    _controller.RegisterCallbackForPinValueChangedEvent(_pin.VSYNC, PinEventTypes.Rising, VSYNC_ValueChanged);
    .......
}

private void VSYNC_ValueChanged(object sender, PinValueChangedEventArgs e)
{
    foreach (var item in _controller._openPins)
    {
        Console.Write($"{item} ");
    }
    Console.WriteLine("-------");

    while (_controller.Read(_pin.HREF) == PinValue.Low) ;    // throw an exception
    .........
}
```

ZhangGaoxing, 2019-08-10:
My visual studio can not attache to the raspberry pi process, so I couldn't debug it and I couldn't find a way to solve the problem.

Frankenslag, 2019-08-11:
@ZhangGaoxing 

I have tried the code below which I home mimics what you are doing... I don't see a problem. What is your hardware and OS ?

    class Program
    {

        static GpioController gpioController;

        const int pinID_VSYNC = 13;
        const int pinID_HREF = 12;

        static void Main(string[] args)
        {
            gpioController = new GpioController(PinNumberingScheme.Logical, new SysFsDriver());

            gpioController.OpenPin(pinID_VSYNC, PinMode.Input);
            gpioController.OpenPin(pinID_HREF, PinMode.Input);

            gpioController.RegisterCallbackForPinValueChangedEvent(pinID_VSYNC, PinEventTypes.Rising, VSYNC_ValueChanged);

            // forever
            while(true){ System.Threading.Tasks.Task.Delay(1000).Wait(); }

        }

        static private void VSYNC_ValueChanged(object sender, PinValueChangedEventArgs e)
        {
            while (gpioController.Read(pinID_HREF) == PinValue.Low) { System.Threading.Tasks.Task.Delay(1000).Wait(); }    // throw an exception
        }
    }

SteveMacha1, 2021-05-29:
I have tried the above sample using Visual Studio 2019  .net C# on a Win10 IOT  machine it seems that registered callbacks are operating properly.  I am using V1.4 of System.Device.Gpio from GitHub.   However when I move the sample into a simple forms application (where the while loop wait delay is not appropriate), then I no longer receive callbacks.   Any thoughts?

Ellerbach, 2021-07-15:
[Triage] @ZhangGaoxing, event handling has been drastically improved since this issue has been open.

@SteveMacha1 if you still have an issue, can you please open a separate issue with more details to reproduce it?

Closing it as no activity for a long time. If still a problem, feel free to reopen it!

closed 2021-07-15 by Ellerbach