dotnet/iot #658: [Question] W3C WebOfThings

tthiery, 2019-08-06:
IOT devices often expose their features into standardized protocols. One of them, pretty new, is [W3C WebOfThings suite](https://www.w3.org/TR/wot-thing-description/) of architecture, descriptions and interfaces standards. Considering we have a world-class HTTP server (speaking of Kestrel) with the necessary feature set (e.g. WebSockets), I wonder if there are any intentions or teams already working on supporting W3C WebOfThings in the .NET IoT space.

I was thinking of writing an implementation of that standard, but the thing is quite huge and I would not mind working in a group on that 😀.

ps: I know this is not the right repo for that, but that is the best entry point I saw so far.

shaggygi, 2019-08-06:
There was an issue created recently somewhat related.

https://github.com/Azure/IoTPlugandPlay/issues/20

krwq, 2019-08-06:
@tthiery I'd be happy if you picked this up with couple of notes.

1. Before we start to do any work let's first discuss the scenarios we want to solve so that we have clear goals.
2. Let's focus on min-value product first: i.e. let's only start with implementing part of the spec which immediately will be useful and for now cut everything else. We need to polish those APIs and make sure they at least partially solve original scenarios.
3. Once we have a proof of concept we can itemize or finish the remainder of the spec.

Does that sound reasonable?

shaggygi, 2019-08-06:
Understanding this issue was more specific to W3C WebOfThings, I believe it would be a good idea to reach out to the [IotPlugAndPlug](https://github.com/Azure/IoTPlugandPlay) team to get a better understanding what their deliverables will be.

As you can see from [this comment](https://github.com/Azure/IoTPlugandPlay/issues/6#issuecomment-489143745), there are SDKs and such already in the works.  Maybe @ChrisGMsft can provide some updates?  After we get a clearer picture of those APIs, we could begin researching how to build upon for bindings, tooling, etc..

IMHO, having closer collaboration between these 2 repos will be a big effort in objectives along with traction in usage for both Azure IoT and System.Device.* APIs.

tthiery, 2019-08-07:
@shaggygi Peeking into the [IotPlugAndPlay](https://github.com/Azure/IoTPlugandPlay) repo, I see an interesting contract. Similar in simplicity what I see in the (early?) Mozilla Draft for the W3C WoT standard. In case I would architect a solution here, I would build a core definition model which can then be represented here as the contract from the Azure group, the W3C one or the (already in use) Mozilla one. And there are surely multiple other. However, the art is to not overengineer this core. Which is tricky once you consider other use cases like local clients for remote Things.

@krwq 
**Scenarios** (and I am not a professional IoT device builder but trying myself in the space of home automation) 

- Build a Thing which I can control via a generic Gateway (like Mozilla's one) or App.
- Build a Thing running on a Linux or Windows (even desktop PC) being controllable by a Thing Gateway (I know, sounds like there are plenty alternatives, however once you consider the app/gateway ecosystem on the other side of the Thing, it is getting interesting).

**Minimal Viable Product** (and I hate them ;))

- A core description model
- A data manager which allows to expose properties, actions, and events (pull/push based)
- An HTTP endpoint exposing the description and value endpoints in W3C WoT (or whatever else is fancy then)

*Note: The not so minimal viable product may include consumer side proxies, proper security and a metadata factory (like `factory.Create("onOffSwitch")`)* 

**Architecture/Delivery**

Achieving a long-term stable API (compared to e.g. the Gpio space) is IMHO not very realistic here. The exposing interfaces are ever evolving around the definition of their core parts (the basic Thing/Property/Action/Event interface is pretty stable when you consider that Mozilla, W3C, Azure and [OpenHAB](https://www.openhab.org/docs/concepts/things.html) all specify it).

ps: I am doing this as a spare time project, so do not expect too much of me here ;)

krwq, 2019-08-07:
@tthiery this sounds reasonable. Could you expand a bit on:

- A core description model
- A data manager which allows to expose properties, actions, and events (pull/push based)

I think we should start with some more or less API shape and how would the code for adding device/thing and APIs for gateway look like (with some example theoretical code) and once we are all happy with that we can proceed with implementing some pieces of it.

For core description model it might be worth to think about merging the effort with https://github.com/dotnet/iot/tree/master/tools/device-listing if possible.

shaggygi, 2019-08-07:
@tthiery 👍 

> An HTTP endpoint exposing the description and value endpoints in W3C WoT (or whatever else is fancy then)

We should try to make it work via different transports (thinking about **Project Bedrock** from the ASP.NET team).  This would allow different connections like Bluetooth, 232/485, ZigBee, etc... in addition to HTTP.

tthiery, 2019-08-07:
@shaggygi Well, once you have a core description model (and the data manager in/out), bindings to different contracts/endpoints are just a matter of implementation. The difficulty is to have a best of breed understanding OR accepting change to the core model over time.

@krwq Regards the code at https://github.com/dotnet/iot/tree/master/tools/device-listing ... I do not think that this is a match. In this WoT/Mozilla/openHAB thinking, an exposed Thing is not a connected sensor. It may, but not necessarily. A switch for a living room light is a *Thing*, the GpioPin and its connected relay powering the lamp is not. A weather station is a *Thing* but not the sensors for humidity, temperature, air pressure and water counting sensors. A weather station is still a *Thing* when it loads its data from the internet (all home automations do exactly that if you do not have physical sensors).

@krwq 
So regards the **description model**: I could imagine something like that

````csharp
var onOffThing = new Thing("living-room", "Living Room Switch", "OnOffSwitch")
                {
                    new ThingProperty<bool>("on", "OnOffProperty"),
                };
````

with that you could then use a **data manager** (for the lack of a better word for now) to host a set of states.

````csharp
var manager = new ThingManager();
manager.Add(onOffThing);

// setting a value for a property (to be exposed to external bindings)
manager.SetPropertyValue<bool>("living-room", "on", true); // variant: setting explicitely
manager.BindPropertyValue<bool>("living-room", "on", () => _globalForCurrentLightStatus); // variant: lookup method

// observing to change applied to the manager (e.g. by an external binding)
manager.OnPropertyChange<bool>("living-room", "on", value => {
    pin.Set(_ledPin, value ? PinValue.High : PinValue.Low);
});
````
Similar patterns for Actions and Events.

*I know that the manager interface is a bit stupid. I did not want to violate the ISP principle on the `Thing` itself where I would mix the domains description and data. The data domain has quite some complexity in it. That interface need definately some refinement.*

I omit here the integration into e.g. **HTTP/WebSockets** considering that this is far off and indeed, there is Project Bedrock and the default ASP.NET Startup style is also well known. In the end you can use JSON against endpoints like
````
PUT /things/living-room/on

true // PUT body; this is the valid JSON for a bool (without a JSON object around it).
````

A weather station would have something like ...
````csharp
var weatherStation = new Thing("weather-station", "Weather Station - Roof", "WeatherStation")
                {
                    new ThingProperty<Weather>("current", "WeatherData"),
                };

// ...

manager.BindPropertyValue<bool>("living-room", "on", () => {
    var (temperature, humidity) = ReadFromTempatureAndHumiditySensors();
    return new Weather(temperature, humidity, null);
);
````

A retrieving call would be ...

````
GET /things/weather-station/current

.. with response serializing the Weather type above into a JSON object ...
{
    "temperature": 22.5,
     "humidity": 45.2,
     "ozonLevel": null
}
````

For the vocabulary "WeatherStation", "WeatherData" or in the above sample "OnOffSwitch" and "OnOffProperty" there are some standards. Have not found the right terminology foundation yet. Apps/Gateways use this terminology to automatically create a suitable UI for it.

krwq, 2019-08-07:
So if things are orthogonal to sensors then would it make sense to make this a separate library/repo in such case?

tthiery, 2019-08-08:
Yes, that is accurate. It is ortogonal to everything currently assembled in this repo. These kind of protocols belong to the heart of an IoT application like the sensors do (and hopefully an embeddable .NET one day ... one can dream).

Also there is no interface between these two domains.

krwq, 2019-08-08:
Let's do following: since you're interested in a subject and it seems useful long-term it might be worth to add it first as a sample project (we have a `sample` directory under the root directory) and then from there we can decide if this is something we want to move inside this repo or if it should stay outside (or perhaps just take some pieces of it and move some pieces out). I.e. you might want to create a weather station as an example with some example code consuming it.

I wouldn't go as far as implementing whole spec just to find out if idea was good though.

Does that sound like a reasonable approach?

tthiery, 2019-08-08:
For now I will prototype some stuff in a separate repo. I will return to you once I have something to show.

krwq, 2019-08-08:
@tthiery we have already started some conversations internally related to these scenarios although not with Azure Plug & Play team

Ellerbach, 2021-07-15:
[Triage] We do have a specific issue tracking the effort on Azure Plug&Play #418 
So closing this one as the other one covers this. Feel free to participate in the other one.

closed 2021-07-15 by Ellerbach