dotnet/iot #661: Raspberry Pi not detected using Ubuntu and Debian

Frankenslag, 2019-08-06:
I believe that I have discovered that the detection logic for determining if a board is a Raspberry Pi only works if running Raspbian and not Ubuntu or Debian.

This is because /proc/cpuinfo doesn't bring back information about the SOC on Debian or Ubuntu.

I believe we should consider improving the detection logic for future releases perhaps using model information from the device tree.

joperezr, 2019-08-06:
The related conversation for this is: https://github.com/dotnet/iot/pull/595#discussion_r310729445

Changes that were prototyped that would fix this: https://github.com/dotnet/iot/commit/376570188c4871aa05d4e5e71143723bb46a7e65#diff-35260effbe2d55bb01d69655a81939e6R13-R43

```c#
            private const string ModelInfoPath = "/proc/device-tree/model";
            //...            

            // a first chance of finding a raspberry pi..... perhaps not on Raspbian note 
            // that we want this to fail quietly if something goes wrong with the test
            try
            {
                if (File.ReadAllText(ModelInfoPath).StartsWith("Raspberry Pi"))
                {
                    return new RaspberryPi3Driver();
                }
            }
            catch { }
```

Frankenslag, 2019-10-03:
@joperezr 

Is now the right time to to work on this again ?

mjjames, 2019-10-03:
I agree, as I have to modify this for my issue #762 I’m happy to do this as part of it just in a separate PR, does it make sense to do so?

krwq, 2019-10-04:
yes, it's ok, we've only locked 3.0 branch

Ellerbach, 2020-12-22:
With the documentation added, some changes in the detection logic, and possibility to use a proper driver, I feel this is addressed now. If there is still something specific, feel free to reopen this issue.

closed 2020-12-22 by Ellerbach